smartdevicelink/sdl_core #3338: Once DeadLock in TranpsortManager during transport disconnect

GetmanetsIrina, 2020-04-16:
### Bug Report
Once DeadLock in TranpsortManager during transport disconnect

##### Reproduction Steps
1. HMI and SDL are started
2. ~3 Mobile app is registered
3. Perform transport disconnect

##### Expected Behavior
SDL continues work without deadlock

##### Observed Behavior
DeadLock occurs  in TransportManager
```
TRACE [13 Apr 2020 11:28:51,937][0x7ff243422700][TransportManager] /home/developer/sdl/repo/SDLMirror/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc:1084 TransportManagerImpl::Handle: enter
DEBUG [13 Apr 2020 11:28:51,937][0x7ff243422700][TransportManager] /home/developer/sdl/repo/SDLMirror/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc:1341 TransportManagerImpl::Handle: Event ON_RECEIVED_FAIL
DEBUG [13 Apr 2020 11:28:51,937][0x7ff243422700][TransportManager] /home/developer/sdl/repo/SDLMirror/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc:1356 TransportManagerImpl::Handle: event_type = ON_RECEIVED_FAIL
TRACE [13 Apr 2020 11:28:51,937][0x7ff243422700][TransportManager] /home/developer/sdl/repo/SDLMirror/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc:1391 TransportManagerImpl::Handle: exit
TRACE [13 Apr 2020 11:28:51,937][0x7ff243422700][TransportManager] /home/developer/sdl/repo/SDLMirror/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc:1084 TransportManagerImpl::Handle: enter
TRACE [13 Apr 2020 11:28:51,938][0x7ff243c23700][TransportManager] /home/developer/sdl/repo/SDLMirror/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc:1401 TransportManagerImpl::Handle: enter
DEBUG [13 Apr 2020 11:28:51,938][0x7ff243c23700][TransportManager] /home/developer/sdl/repo/SDLMirror/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc:1424 TransportManagerImpl::Handle: Got adapter 0x5615372938a0[0] by session id 36
ERROR [13 Apr 2020 11:28:51,938][0x7ff243c23700][TransportManager] /home/developer/sdl/repo/SDLMirror/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc:1438 TransportManagerImpl::Handle: Data sent error
TRACE [13 Apr 2020 11:28:51,938][0x7ff243c23700][TransportManager] /home/developer/sdl/repo/SDLMirror/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc:1444 TransportManagerImpl::Handle: exit
```

##### Note: 
1. It is unable to reproduce issue on previous release because of https://github.com/smartdevicelink/sdl_core/issues/3149, but according to code analysis this is not regression issue and presents on previous releases.
2. The issue can be reproduced more often after fixing of #3323

##### OS & Version Information
* sdl_hmi branch: [develop](https://github.com/smartdevicelink/sdl_hmi/tree/develop)
* commit: 74d24927cf77282c44678eba219adadbb6dcd072
* sdl_core branch: https://github.com/smartdevicelink/sdl_core/tree/6.1.0_RC
* transport: USB

##### Logs
[SmartDeviceLinkCore.zip](https://github.com/smartdevicelink/sdl_core/files/4485782/SmartDeviceLinkCore.zip)

GetmanetsIrina, 2020-04-28:
Issues is reproduced on release_6.1_RC_2, issues has occurrences 1/100.

AKalinich-Luxoft, 2020-06-23:
@theresalech PR with fix is ready for Livio review.

theresalech, 2020-06-23:
Hi @AKalinich-Luxoft, this issue is not currently planned for the Core 6.2 release, but we will work to include if time allows.

jacobkeeler, 2020-07-29:
Closing with the merge of #3347

PR #3347 (2020-07-29): Fix mutex deadlock around app button subscriptions
Fixes #3338 

This PR is **not ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by manual testing

### Summary
Sometimes there was observed mutex deadlock while more then one thread are accessing buttons subscriptions container protected with data accessor class. The issue is happening when one thread is trying to unregister application, acquires applications lock and trying to save application resumption data which at some point requires app subscribed buttons lock to be acquired. At the same moment an [... cut]
closed 2020-07-29 by jacobkeeler