smartdevicelink/sdl_core #3340: Build fails for EXTERNAL_PROPRIETARY policy mode

dboltovskyi, 2020-04-21:
### Bug Report
Build fails for EXTERNAL_PROPRIETARY policy mode

##### Reproduction Steps
1. Try to build SDL with `-DEXTENDED_POLICY=EXTERNAL_PROPRIETARY` cmake option

##### Expected Behavior
Build finished successfully

##### Observed Behavior
Build fails with error:
```
[ 24%] Linking CXX executable policyValidator
In file included from /home/developer/sdl/sdl_core/src/components/application_manager/include/application_manager/message_helper.h:46:0,
                 from /home/developer/sdl/sdl_core/src/components/application_manager/src/message_helper/message_helper.cc:36:
/home/developer/sdl/sdl_core/src/components/include/application_manager/policies/policy_handler_interface.h:354:13: error: ‘PTUIterationType’ does not name a type
       const PTUIterationType iteration_type =
```

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/6.1.0 (ab90f16653)
* Testing Against: N/A

##### Test Case, Sample Code, and / or Example App
N/A

dboltovskyi, 2020-04-21:
@theresalech Please triage this regression issue.

theresalech, 2020-04-21:
@dboltovskyi thank you for reporting.  Can you please advise if the Luxoft team plans to submit a PR to resolve this issue?

dboltovskyi, 2020-04-21:
@theresalech We appreciate if PM fixes this issue. Thank you.

jacobkeeler, 2020-04-21:
Closing with the merge of #3341

PR #3341 (2020-04-21): Move cached app logic to ChoosePTUApplication

Fixes #3340 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Build with EXTERNAL_PROPRIETARY policy mode

### Summary
Properly separate PROPRIETARY/HTTP retry logic from EXTERNAL_PROPRIETARY build

### Changelog

##### Bug Fixes
* Fix EXTERNAL_PROPRIETARY policy build

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-04-21 by jacobkeeler