smartdevicelink/sdl_core #3346: Cyclic UpdateDeviceList UpdateAppList in case device is connected or app is registered and BT is turned On

GetmanetsIrina, 2020-04-28:
### Bug Report
Cyclic UpdateDeviceList UpdateAppList in case device is connected or app is registered and BT is turned On

##### Reproduction Steps
1. BT is turned ON (before SDL start!)
2. SDL and HMI are started
3. Connect USB device or register app via WiFi

##### Expected Behavior
SDL sends BC.UpdateAppList in case of app registration of unregistration and  BC.UpdateDeviceList after device connection and disconnection.

##### Observed Behavior
SDL sends BC.UpdateDeviceList and BC.UpdateAppList with the same data each ~50 seconds

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/6.1.0_RC2
* HMI:https://github.com/smartdevicelink/sdl_hmi/commit/7a7e6acddc12b8b4e9d2b98df9c09ee423f6356d

##### Note
This is not regression issue, issue is reproduced on release 6.0

##### Logs
[SmartDeviceLinkCore_Update_App_Device_List.zip](https://github.com/smartdevicelink/sdl_core/files/4545956/SmartDeviceLinkCore_Update_App_Device_List.zip)

GetmanetsIrina, 2020-04-28:
This is not the regression issue, issue is reproduced on release 6.0

theresalech, 2021-11-10:
@GetmanetsIrina can you please advise if Luxoft is planning to submit a PR to resolve this issue? Thank you!

dboltovskyi, 2021-11-10:
@theresalech Yes, Luxoft team is planning to submit a fix for this issue.

jacobkeeler, 2022-01-13:
Closed via #3835

closed 2022-01-13 by jacobkeeler