smartdevicelink/sdl_core #3364: App freezes on HMI after transport disconnect

GetmanetsIrina, 2020-05-13:
### Bug Report
App freezes on HMI after transport disconnect

##### Occurrences
1/10

##### Reproduction Steps
1. Several (navi and media) apps are registered
2. Some app is activated
3. Perform transport reconnect

##### Expected Behavior
All registered apps are unregistered during transport disconnect and successfully registered after transport connect.

##### Observed Behavior
One app freezes on HMI after transport disconnect and SDL responds to RAI request for frozen app with `APPLICATION_REGISTERED_ALREADY` resultCode after transport connect

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commits/develop
* Transport: USB

##### Logs
[ApplicationRegisterAlready.zip](https://github.com/smartdevicelink/sdl_core/files/4621120/ApplicationRegisterAlready.zip)

chinlin0924, 2020-05-14:
We got the same issue too. This is very easy to reproduce if you perform USB connect/disconnect for long time.

VjKlepikov, 2021-11-08:
Issue is not reproduced on the latest develop [95512d7b](https://github.com/smartdevicelink/sdl_core/commits/develop)

closed 2021-11-08 by VjKlepikov