smartdevicelink/sdl_core #3384: Processing of individual vehicle data result codes in SubscribeVehicleData/UnsunscribeVehicleData responses from HMI

GetmanetsIrina, 2020-05-20:
### Bug Report
Processing of individual vehicle data result codes in SubscribeVehicleData/UnsunscribeVehicleData responses from HMI 

##### Reproduction Steps
1. SDL and HMI are started
2. `SubscribeVehicleData`,`UnsunscribeVehicleData` are allowed by policies
3. App is subscribed to e.g speed data
4. App requests `SubscribeVehicleData(gps)`
5. HMI responds with `SubscribeVehicleData(resultCode:SUCCESS, gps = { resultCode = <not success code from VehicleDataResultCode structure> })`
6. App requests `UnsubscribeVehicleData(speed)`
7.  HMI responds with `UnsubscribeVehicleData(resultCode:SUCCESS, gps = { resultCode = <not success code from VehicleDataResultCode structure> })`

##### Expected Behavior
TBA

##### Observed Behavior
1. SDL sends `SubscribeVehicleData(resultCode:SUCCESS, gps = { resultCode = <not success code from VehicleDataResultCode structure> })` response to mobile app and subscription is not created internally
2. SDL sends `UnsubscribeVehicleData(resultCode:GENERIC_ERROR, gps = { resultCode = <not success code from VehicleDataResultCode structure> }, info = "Unsubscription failed for some Vehicle data")` response to mobile app and unsubscription is failed internally

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/bb4a6fb4d44bf86bd66d4827718d393244b8078e

##### Test Case, Sample Code, and / or Example App
[SubscribeVehicleData_SmartDeviceLinkCore.zip](https://github.com/smartdevicelink/sdl_core/files/4655985/SubscribeVehicleData_SmartDeviceLinkCore.zip)
[UnsubscribeVehicleData_SmartDeviceLinkCore.zip](https://github.com/smartdevicelink/sdl_core/files/4655986/UnsubscribeVehicleData_SmartDeviceLinkCore.zip)

ShobhitAd, 2022-02-09:
@GetmanetsIrina Is this issue still reproducible on the latest develop commit

GetmanetsIrina, 2022-02-10:
@ShobhitAd, yes, the issue is still reproducible on the latest develop commit (2f002290411cae9a414c4dfccc93bb777a9482d0)