smartdevicelink/sdl_core #3396: SDL doesn't start PTU on Days, Ignition Cycles or new Ignition cycle triggers for HTTP policy mode

dboltovskyi, 2020-05-25:
### Bug Report
After merge of https://github.com/smartdevicelink/sdl_core/pull/3363 SDL does not start PTU sequence for the following triggers in `HTTP` policy mode:
 `1)` Number of ignition cycles (`exchange_after_x_ignition_cycles`)
 `2)` Number of days (`exchange_after_x_days`)
 `3)` New ignition cycle after unsuccessful retry

##### Reproduction Steps
1. Make sure SDL is built in `HTTP` policy mode
2. Start SDL, HMI, connect Mobile device, register application
3. Make sure at least one PTU trigger is in place: `1)`, `2)` or `3)`

##### Expected Behavior
SDL start PTU sequence by sending messages:
 - to HMI: SDL.OnStatusUpdate(UPDATE_NEEDED, UPDATING)
 - to App: OnSystemRequest

##### Observed Behavior
SDL does not start PTU sequence

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (b982ad29b9)
* Testing Against: ATF:
  - [052_ATF_PTU_Trigger_IGN_Cycles_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/build_options/052_ATF_PTU_Trigger_IGN_Cycles_HTTP.lua)
  - [053_ATF_Policy_Table_Update_Trigger_After_N_Days_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/build_options/053_ATF_Policy_Table_Update_Trigger_After_N_Days_HTTP.lua)
  - [098_ATF_PTU_Unsuccessful_Even_After_Retry_Strategy_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/build_options/098_ATF_PTU_Unsuccessful_Even_After_Retry_Strategy_HTTP.lua)

##### Test Case, Sample Code, and / or Example App
 - [052.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4678516/052.tar.gz)
 - [053.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4678520/053.tar.gz)
 - [098.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4678522/098.tar.gz)

dboltovskyi, 2020-05-25:
@theresalech Please triage this issue since it's regression one

dboltovskyi, 2020-05-25:
Issue is also applicable for new [028_PTU_snapshot_validation_no_app_connected.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/WebEngine/028_PTU_snapshot_validation_no_app_connected.lua) script for WebEngine.
Script is aborted since PTS was not created.
Logs: [028.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4678558/028.tar.gz)

theresalech, 2020-05-26:
@dboltovskyi will Luxoft be submitting a PR to resolve this issue?

dboltovskyi, 2020-05-26:
@theresalech We will appreciate if PMs be able to submit a fix.

iCollin, 2020-06-05:
Closed via #3415

PR #3415 (2020-06-05): trigger ptu on startup only in proprietary policy mode
Fixes #3396

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF

### Summary
only "triggerPTUonStartup" in proprietary mode

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-06-05 by iCollin