smartdevicelink/sdl_core #3399: Unit test in WebsocketSessionTest fails with core dump

ychernysheva, 2020-05-26:
### Bug Report
Currently on develop branch unit test WebsocketSessionTest._SendMessage_UnpreparedConnection_WithoutFall fails with ENABLE_GCOV=ON flag (on any policy flow).
##### Reproduction Steps
1. Build SDL core with flag ENABLE_GCOV=ON
2. Run command "make test"

##### Expected Behavior
No core dump

##### Observed Behavior

WebsocketSessionTest.SendMessage_UnpreparedConnection_WithoutFall
/home/developer/sdl/sdl_core/src/components/hmi_message_handler/test/hmi_message_handler_impl_test.cc:230: Failure
Death test: send_message()
    Result: died but not with expected exit code:
            Terminated by signal 11 (core dumped)
Actual msg:
[  DEATH   ] 
[  FAILED  ] WebsocketSessionTest.SendMessage_UnpreparedConnection_WithoutFall (4 ms)

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: 6.1.0
* gcc compiler version: 7.4

iCollin, 2020-05-27:
Hello @ychernysheva does your team plan to make a fix or should PM investigate this issue?

ychernysheva, 2020-05-28:
@iCollin Yes, I've already prepared potential fix for this issue, now this PR is on internal review.

iCollin, 2020-06-04:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3418

closed 2020-06-04 by iCollin