smartdevicelink/sdl_core #3402: Report on HMI_API.xml issues(inaccurate description)

Yuki-Shoda-Nexty, 2020-05-29:
### Report on HMI_API.xml issues(inaccurate description)

We suggest correcting HMI_API.xml(inaccurate description)

In the current `HMI_API.xml`, there is a problem of lack of description in the parameter <description />.
By compensating for this, we can provide a more accurate RPC documents to the SDL developers.
The target items are described in the Excel file.

[HMI_API_inaccurate description.xlsx](https://github.com/smartdevicelink/sdl_core/files/4699385/HMI_API_inaccurate.description.xlsx)

**e.g.** Some HMILevel elements are missing in HMI_API.xml. Add the description of each element as shown below.

Modify the description of `alignment` as shown below.

```diff
     <param name="alignment" type="TextAlignment" mandatory="false" since="1.0">
       <description>
-        Specifies how mainField1 and mainField2 texts should be aligned on display.
+        Specifies how mainField1, mainField2, mainField3 and mainField4 texts should be aligned on display.
         .If omitted, texts will be centered.
       </description>
     </param>
```

Yuki-Shoda-Nexty, 2020-05-29:
@theresalech -san
Would this be a good way to report document problems?

theresalech, 2020-06-02:
Hi @Yuki-Shoda-Nexty, yes this is a good way to report issues with the HMI_API.xml.  Can you please confirm that this issue contains all inaccurate descriptions within the HMI_API.xml?  If so, please submit a PR to fix the descriptions/resolve the issue, and the PM can review.  Thank you!

Yuki-Shoda-Nexty, 2020-06-04:
@theresalech -san, Fixed description and example in the Bug Report.

theresalech, 2020-06-04:
@Yuki-Shoda-Nexty, thank you!  Please advise when a PR has been entered to resolve the issue

Yuki-Shoda-Nexty, 2020-06-08:
@theresalech -san
We have already finished confirmation of the issues you indicated.
We do not propose it as PR, but we will report it on the issue page instead.
We would like PM to fix the issue by using the information on issue page as reference.

theresalech, 2020-06-08:
@Yuki-Shoda-Nexty understood, thank you.  This issue has not been raised as high priority for the upcoming release by any Steering Committee members, but the Project Maintainer will work to resolve the issue as time allows.

Yuki-Shoda-Nexty, 2020-06-09:
@theresalech -san, Thank you. Please comment if you have any questions.