smartdevicelink/sdl_core #3410: CCB :  Crash related to HandleOnEvent

LuxoftAKutsan, 2020-05-29:
### Bug Report
 Crash related to HandleOnEvent  

##### Reproduction Steps
N\A

##### Expected Behavior
No crash 

##### Observed Behavior
sdl crash 

##### OS & Version Information
##### OS & Version Information
* OS/Version: QNX 7.0 
* SDL Core Version: [5.0.1]( https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1) integrated to SYNC environment   
* Testing Against: Hardware Revision: e.g. EV1, EV2, EV2.1, DV1, PV1, etc.

LitvinenkoIra, 2021-01-18:
Should be fixed after https://github.com/smartdevicelink/sdl_core/issues/3141

VladSemenyuk, 2021-01-26:
Should be fixed after https://github.com/smartdevicelink/sdl_core/pull/3591 merge by moving https://github.com/CustomSDL/Sync3.2v2/pull/780 pull request from Sync3

VladSemenyuk, 2021-01-26:
Should be fixed after https://github.com/smartdevicelink/sdl_core/pull/3591 merge