smartdevicelink/sdl_core #3414: Build fails for HTTP policy mode with UT

dboltovskyi, 2020-06-01:
### Bug Report
SDL build fails for HTTP policy mode with UT

##### Reproduction Steps
1. Try to build SDL with the following CMake build flags:
```
-DBUILD_TESTS=ON -DEXTENDED_POLICY=HTTP
```

##### Expected Behavior
Build finished successfully

##### Observed Behavior
Build failed with error:

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (c813503)
* Testing Against: N/A

##### Test Case, Sample Code, and / or Example App
  - Build log:
```
18:15:55 [ 96%] Building CXX object src/components/application_manager/rpc_plugins/app_service_rpc_plugin/CMakeFiles/app_service_rpc_plugin_static.dir/src/app_service_app_extension.cc.o
18:15:56 In file included from /home/jenkins/workspace/Develop_SDL_UT_H/src/3rd_party-static/gmock-1.7.0/include/gmock/gmock-generated-function-mockers.h:43:0,
18:15:56                  from /home/jenkins/workspace/Develop_SDL_UT_H/src/3rd_party-static/gmock-1.7.0/include/gmock/gmock.h:61,
18:15:56                  from /home/jenkins/workspace/Develop_SDL_UT_H/src/components/application_manager/test/policy_handler_test.cc:37:
18:15:56 /home/jenkins/workspace/Develop_SDL_UT_H/src/components/application_manager/test/policy_handler_test.cc: In member function ‘virtual void test::components::policy_handler_test::PolicyHandlerTest_DISABLED_OnSnapshotCreated_UrlAdded_Test::TestBody()’:
18:15:56 /home/jenkins/workspace/Develop_SDL_UT_H/src/components/application_manager/test/policy_handler_test.cc:1862:78: error: call of overloaded ‘gmock_SendPolicySnapshotNotification(const testing::internal::AnythingMatcher&, const testing::internal::AnythingMatcher&, const testing::internal::AnythingMatcher&, const testing::internal::AnythingMatcher&)’ is ambiguous
18:15:56    EXPECT_CALL(mock_message_helper_, SendPolicySnapshotNotification(_, _, _, _));
18:15:56                                                                               ^
18:15:56 /home/jenkins/workspace/Develop_SDL_UT_H/src/3rd_party-static/gmock-1.7.0/include/gmock/gmock-spec-builders.h:1817:20: note: in definition of macro ‘GMOCK_EXPECT_CALL_IMPL_’
18:15:56      ((obj).gmock_##call).InternalExpectedAt(__FILE__, __LINE__, #obj, #call)
```

dboltovskyi, 2020-06-03:
@theresalech Please triage this regression issue

theresalech, 2020-06-09:
Closed via #3422

PR #3422 (2020-06-09): fix build error in http unit tests
Fixes #3414

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
build core with `-DBUILD_TESTS=ON -DEXTENDED_POLICY=HTTP`

### Summary
fix compile error: ambiguous call

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-06-09 by theresalech