smartdevicelink/sdl_core #3419: [SDL 0305] Homogenize TextFieldName

theresalech, 2020-06-03:
Proposal: [Homogenize TextFieldName](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0305-homogenize-textfieldname.md)

> This proposal is to homogenize the enum `TextFieldName` in [`MOBILE_API.xml`](https://github.com/smartdevicelink/rpc_spec/blob/RPC-Generator/MOBILE_API.xml#L753) and [`HMI_API.xml`](https://github.com/smartdevicelink/sdl_core/blob/6.1.1/src/components/interfaces/HMI_API.xml#L539).

Review: https://github.com/smartdevicelink/sdl_evolution/issues/1020

Steering Committee Decision:

>The Steering Committee voted to accept this proposal with the following revision: add requirement for documentation of enum values, as described in [this comment](https://github.com/smartdevicelink/sdl_evolution/issues/1020#issuecomment-637740732).

> The proposal .md file was updated to reflect these revisions on 6/3/2020.

jordynmackool, 2021-01-27:
The proposal markdown file has been updated per the revisions included in the accepted with revisions review issue: [Revise SDL-0305 Homogenize TextFieldName](https://github.com/smartdevicelink/sdl_evolution/issues/1113). 

Accepted revisions can be reviewed in the merged PR here: https://github.com/smartdevicelink/sdl_evolution/pull/1109.

iCollin, 2021-02-09:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3595

PR #3595 (2021-02-09): SDL-0305 Homogenize TextFieldName
Fixes #3419

This PR is **not ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan

### Summary
Implements proposal [SDL 0305](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0305-homogenize-textfieldname.md) which Homogenizes the Enum TextFieldName in HMI_API.xml with MOBILE_API.xml in the Rpc spec. Removing `notificationText`, adding descriptions to`timeToDestination` and `turnText` for enum `TextFieldName`

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zH [... cut]
closed 2021-02-09 by iCollin