smartdevicelink/sdl_core #3421: SSL Context should use MACRO instead of direct struct access

mrapitis, 2020-06-05:
### Bug Report
SDL Core will not build with security enabled when using OpenSSL 1.1 because the SSL context used inside of ResetConnection is directly accessing the underlying structure inside of the connection.  Instead the context should be referenced by the proper MACRO which will allow core to build with OpenSSL 1.1 as well as older versions.

Replacing the line linked below with the following code snippet resolves this issue
https://github.com/smartdevicelink/sdl_core/blob/master/src/components/security_manager/src/ssl_context_impl.cc#L599
```SSL_CTX* ssl_context = SSL_get_SSL_CTX(connection_);```

##### Reproduction Steps
Clone SDL Core and attempt to build on a system with security enabled while using OpenSSL 1.1

##### Expected Behavior
Build is performed successfully

##### Observed Behavior
Error when attempting to build

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: 6.1.1

dboltovskyi, 2020-06-05:
@mrapitis Please notice this issue may be related to https://github.com/smartdevicelink/sdl_core/issues/3220

iCollin, 2021-01-21:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3582

PR #3582 (2021-01-21): Fix getting SSL context in CryptoManager
Fixes #3421 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by unit tests

### Summary
This PR fixes way of getting SSL context from SSL connection. Direct access works well in 1.0.2 version, however this field
became unavailable in 1.1.1 version which causes SDL build to fail. In order to make this change working on all lib versions
as expected, direct access has been replaced with convenient getter function from the same library.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/for [... cut]
closed 2021-01-21 by iCollin