smartdevicelink/sdl_core #3423: App icon failed to update due to special characters.

zhouxin627, 2020-06-08:
### Bug Report
App icon failed to update due to special characters.

##### Preconditions
TestApp's name is "Name Name".

##### Reproduction Steps
1.Register the TestApp.
2.Start the TestAPP.
3.After sending `PutFile(name name)` and receiving `PutFileResponse(success)`, send `setAppIcon(name name)`.
4.Confirm AppIcon's update.

##### Expected Behavior
App icon updated successfully

##### Observed Behavior
App icon failed to update and the log shows no file named `name%20name`.

##### Sequence Diagram
[image]

##### Recommended solution
1.Remove the escape processing(ConvertPathForURL) in SDLCore.
2.HMI side judges whether to carry out the escape processing according to the actual situation.

##### OS & Version Information
* OS/Version: Android 9
* SDL Core Version: 4.11.0
* Testing Against: SDL Core 6.1.0 + HMI running on our development environment.

##### Occurrence:
100%

##### Recovery Method:
None

zhouxin627, 2020-09-02:
Hi @theresalech，do you have a plan to resolve this issue.

theresalech, 2020-09-02:
Hi @zhouxin627, the Steering Committee did not prioritize this issue for the Core 7.0 release. The PM will work to resolve as time allows, but we would recommend raising this issue to the Steering Committee to prioritize for the next release if this is priority for Nexty. If you are able to submit a PR to resolve this issue, that's always helpful in expediting an issue.

Shohei-Kawano, 2020-09-16:
@theresalech -san,
Thank you for comment. 
This time it wasn't a request to put it in 7.0, but I didn't know when it would be contributed and reviewed, so we hurriedly requested it. Now that we understand your situation, I would appreciate it if you could contribute and review as soon as possible you have time.

theresalech, 2020-09-16:
@Shohei-Kawano, yes we will work to investigate the issue and submit a PR as time allows, but given set priorities and current time constraints, it is unlikely this will be included in the Core 7.0 release. I'd encourage Nexty to raise this issue as priority to the Steering Committee for the next Core release to better ensure its inclusion there. Thanks!

VNikiforov-dev, 2021-11-15:
Issue is not reproduced at:
- SDL core: https://github.com/smartdevicelink/sdl_core/releases/tag/8.0.0-RC2 (a0f0040)
- HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop (d5e648a)
- APP: SPT 20210729-Android

jacobkeeler, 2022-01-19:
Duplicate of https://github.com/smartdevicelink/sdl_core/issues/3406, which was fixed with https://github.com/smartdevicelink/sdl_core/pull/3407

closed 2022-01-19 by jacobkeeler