smartdevicelink/sdl_core #3460: SDL doesn't provide individual VD result code for clusterModeStatus in SubscribeVehicleData/UnsunscribeVehicleData responses from HMI

dboltovskyi, 2020-07-15:
### Bug Report
SDL doesn't provide individual VD result code for `clusterModeStatus` in `SubscribeVehicleData/UnsunscribeVehicleData` responses from HMI in case of 2 apps.

##### Reproduction Steps
*Preconditions:*
1. SDL and HMI are started
2. `SubscribeVehicleData` RPC and `<vd_param>` parameter are allowed by policies
3. `App_1` and `App_2` are registered
4. `App_1` successfully subscribes to `<vd_param>`

*In case:*
1. `App_2` sends valid `SubscribeVehicleData(<vd_param>=true)` request to SDL

##### Expected Behavior
SDL does:
- send SubscribeVehicleData response with `(success = true, resultCode = "SUCCESS",
 <vd_param> = <data received from HMI>)` to `App_2`
- not transfer this request to HMI

##### Observed Behavior
SDL does:
- not send `<vd_param> = <data received from HMI>)` in response to `App_2`

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (81fb32e)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF and SDL logs:

dboltovskyi, 2020-07-15:
Please take into account that for `SubscribeVehicleData/UnsunscribeVehicleData` real VD parameter name is `clusterModes` instead of `clusterModeStatus`
This discrepancy was described in issue https://github.com/smartdevicelink/sdl_core/issues/1696

theresalech, 2021-11-10:
@dboltovskyi can you please advise if Luxoft is planning to submit a PR to resolve this issue? Thank you!

dboltovskyi, 2021-11-10:
@theresalech Yes, Luxoft team is planning to submit a fix for this issue.

ShobhitAd, 2022-01-07:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3830

VjKlepikov, 2022-01-19:
The issue is reproduced on the current develop branch [8f1ccb2](https://github.com/smartdevicelink/sdl_core/tree/develop) for new scripts PR [2618](https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2618)
Logs: [2022-01-17_08-33-54.236_failed.tar.gz](https://github.com/smartdevicelink/sdl_core/files/7896734/2022-01-17_08-33-54.236_failed.tar.gz)

jacobkeeler, 2022-01-19:
@VjKlepikov It looks like the failure in https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2618 is slightly different than the described issue, but similar. In this case, if the app requests `clusterModeStatus` in `SubscribeVehicleData` (along with other parameters) when it does not have permissions for `clusterModeStatus`. Core should send `clusterModes` in the response with the `DISALLOWED` result code, but instead it omits the `clusterModes` parameter.

ShobhitAd, 2022-02-01:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3847

PR #3830 (2022-01-07): Replace clusterModeStatus with clusterModes and vise versa
Fixes #3460

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF test scripts

### Summary
There is a discrepancy between request and response names of VD parameter with data type `VEHICLEDATA_CLUSTERMODESTATUS`: according to API we should use name `clusterModeStatus` in mobile and HMI requests, but another name `clusterModes` is provided in mobile and HMI responses. To deal with this issue we need to convert request to response names and vise versa to ensure correct processing  [... cut]
closed 2022-02-01 by ShobhitAd