smartdevicelink/sdl_core #3465: [SDL 0296] Possibility to update video streaming capabilities during ignition cycle

theresalech, 2020-07-27:
Proposal: [Possibility to update video streaming capabilities during ignition cycle](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0296-Update-video-streaming-capabilities-during-ignition-cycle.md)

> The main purpose of this proposal is to make it possible to update video streaming capabilities (VSC) during the ignition cycle.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/983

**Steering Committee Decision:**

The Steering Committee voted to accept this proposal with the following revisions:

- (Item 3) For iOS, change `streamingVideoDelegate` to `delegate`.  For Both iOS and Java Suite, use `VideoStreamingRange`.
- (Item 10) Use `VideoStreamingCapability` struct for the AppCapability feature.  While each capability will have to be reviewed to see if they can be reused for both system and app capabilities, we do feel confident enough to reuse `VideoStreamingCapability` for both.
- (Item 11) Add the following to the HMI Integration Guidelines, and extend the `Resolution Switching` diagram to document this behavior.
> The HMI should stop displaying video to the user after resizing the window instead of waiting for the StopStream request from Core. Video would resume after the video service is restarted. See the following example state flow: "user attempts to resize" -> "HMI stops displaying video" -> "head unit sends OnAppCapabilityUpdated" -> "HMI resizes/waits for stop and start of video stream" -> "display new video stream".
- (Item 12) Add `HMI` as an impacted platform.

The proposal .md file was updated to reflect these revisions on 7/27/2020.

theresalech, 2020-08-05:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0296 Revisions - Possibility to update video streaming capabilities during ignition cycle](https://github.com/smartdevicelink/sdl_evolution/issues/1059).  Revisions include updating public API properties for Android to align with iOS, and extending the list of impacted projects to include RPC Spec and JavaScript Suite, as the Mobile API is changed.  Please reference the issue and proposal markdown file for revisions details.

ShobhitAd, 2021-02-12:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3468

closed 2021-02-12 by ShobhitAd

jordynmackool, 2021-03-03:
The proposal markdown file has been updated per the revisions included in the accepted review issue: [Revise SDL-0296 Possibility to update video streaming capabilities during ignition cycle](https://github.com/smartdevicelink/sdl_evolution/issues/1125).

Accepted revisions can be reviewed in the merged PR here: https://github.com/smartdevicelink/sdl_evolution/pull/1124