smartdevicelink/sdl_core #3469: SDL doesn't send RC.SetGlobalProperties during resumption

dboltovskyi, 2020-07-31:
### Bug Report
SDL doesn't send RC.SetGlobalProperties during resumption

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile device, register application
2. Send `SetGlobalProperties` with `userLocation` parameter
=> SDL sends `RC.SetGlobalProperties` request to HMI and after receiving succesful response sends `OnHashChange` notification to the app
3. Perform unexpected disconnect and then reconnect (with correct hashId)
=> SDL starts resumption procedure

##### Expected Behavior
SDL sends RC.SetGlobalProperties during resumption trying to restore global properties

##### Observed Behavior
SDL doesn't send RC.SetGlobalProperties during resumption

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (6560d02)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

LitvinenkoIra, 2020-09-11:
Please notice issue is also applicable for 6.1.1 Release

dboltovskyi, 2020-12-14:
It looks like issue is also applicable for `UI.SetGlobalProperties`
E.g.: if App defined keyboard properties:
```
  keyboardProperties = {
    language = "EN-US",
    keyboardLayout = "AZERTY",
    keypressMode = "SINGLE_KEYPRESS"
```
And then disconnects/reconnects SDL won't send `UI.SetGlobalProperties` during resumption.
Please see: [2020-12-14_14-28-12.786_FAIL.tar.gz](https://github.com/smartdevicelink/sdl_core/files/5691142/2020-12-14_14-28-12.786_FAIL.tar.gz)

However if App provides 2 additional parameters `vrHelpTitle` and `vrHelp` along with the previous ones SDL sends `UI.SetGlobalProperties` during resumption correctly with all required data.
Please see: [2020-12-14_14-29-59.771_SUCCESS.tar.gz](https://github.com/smartdevicelink/sdl_core/files/5691151/2020-12-14_14-29-59.771_SUCCESS.tar.gz)

iCollin, 2021-02-03:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3622

PR #3622 (2021-02-03): Fix RC.SetGlobalProperties during resumption
Fixes #3469 

This PR is ready for review.

Risk
This PR makes no API changes.

Testing Plan
ATF

Summary
Added request for RC_SetGlobalProperties to MessageHelper::CreateGlobalPropertiesRequestsToHMI()
Also UserLocation information added to resumption data (json and DB)

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2021-02-03 by iCollin