smartdevicelink/sdl_core #3482: Memory leaking while video streaming

AlbertGRMN, 2020-08-21:
### Bug Report
Memory leak while video streaming.

##### Reproduction Steps
1. Connect SDL navigation streaming app.
2. Observe the memory usage while video streaming is performed.
3. Terminate SDL Core and check the leaked information from GCC leak sanitizer reports.

##### Expected Behavior
Memory should not constant leak.

##### Observed Behavior
Will leak about 0.3% per hour on a 1G Ram machine and will trigger OOM to get service being killed.

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: 4.5.2

##### Test Case, Sample Code, and / or Example App
**GCC Sanitizer logs:**

_(Leak Type-1)_
Indirect leak of 2800 byte(s) in 10 object(s) allocated from:
    #0 0x7f1342617532 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x99532)
    #1 0x28700b8 in threads::CreateThread(char const*, threads::ThreadDelegate*) /ssd_aoem2/sdlcore/sdl_core441/src/components/utils/src/threads/thread_posix.cc:297
    #2 0x2846b7a in timer::Timer::Timer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, timer::TimerTask*) /ssd_aoem2/sdlcore/sdl_core441/src/components/utils/src/timer.cc:51
    #3 0x19265e3 in app_launch::Launcher::Launcher(resumption::ResumeCtrl const&, app_launch::DeviceAppsLauncher&, app_launch::AppsLauncher&) /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/app_launch/device_apps_launcher.cc:34
    #4 0x1929736 in utils::SharedPtr<app_launch::Launcher> utils::MakeShared<app_launch::Launcher, resumption::ResumeCtrl, app_launch::DeviceAppsLauncher, app_launch::AppsLauncher>(resumption::ResumeCtrl&, app_launch::DeviceAppsLauncher&, app_launch::AppsLauncher&) /ssd_aoem2/sdlcore/sdl_core441/src/components/include/utils/make_shared.h:88
    #5 0x192790f in app_launch::LauncherGenerator::operator()() const /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/app_launch/device_apps_launcher.cc:111
    #6 0x1929c47 in std::back_insert_iterator<std::vector<utils::SharedPtr<app_launch::Launcher>, std::allocator<utils::SharedPtr<app_launch::Launcher> > > > std::generate_n<std::back_insert_iterator<std::vector<utils::SharedPtr<app_launch::Launcher>, std::allocator<utils::SharedPtr<app_launch::Launcher> > > >, unsigned short, app_launch::LauncherGenerator>(std::back_insert_iterator<std::vector<utils::SharedPtr<app_launch::Launcher>, std::allocator<utils::SharedPtr<app_launch::Launcher> > > >, unsigned short, app_launch::LauncherGenerator) /usr/include/c++/5/bits/stl_algo.h:4337
    #7 0x1927c7b in app_launch::DeviceAppsLauncherImpl::DeviceAppsLauncherImpl(app_launch::DeviceAppsLauncher&, app_launch::AppsLauncher&) /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/app_launch/device_apps_launcher.cc:130
    #8 0x1925db9 in app_launch::DeviceAppsLauncher::DeviceAppsLauncher(application_manager::ApplicationManager&, app_launch::AppsLauncher&, app_launch::AppLaunchSettings const&) /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/app_launch/device_apps_launcher.cc:197
    #9 0x191f448 in app_launch::AppLaunchCtrlImpl::AppLaunchCtrlImpl(app_launch::AppLaunchData&, application_manager::ApplicationManager&, app_launch::AppLaunchSettings const&) /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/app_launch/app_launch_ctrl_impl.cc:57
    #10 0x1857b13 in application_manager::ApplicationManagerImpl::Init(resumption::LastState&, media_manager::MediaManager*) /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/application_manager_impl.cc:2236
    #11 0x17ead91 in main_namespace::LifeCycle::StartComponents() /ssd_aoem2/sdlcore/sdl_core441/src/appMain/life_cycle.cc:116
    #12 0x17fd7c4 in main /ssd_aoem2/sdlcore/sdl_core441/src/appMain/main.cc:137
    #13 0x7f134033082f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)

_(Leak Type-2)_
Indirect leak of 2720 byte(s) in 10 object(s) allocated from:
    #0 0x7f1342617832 in operator new(unsigned long, std::nothrow_t const&) (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x99832)
    #1 0x1929717 in utils::SharedPtr<app_launch::Launcher> utils::MakeShared<app_launch::Launcher, resumption::ResumeCtrl, app_launch::DeviceAppsLauncher, app_launch::AppsLauncher>(resumption::ResumeCtrl&, app_launch::DeviceAppsLauncher&, app_launch::AppsLauncher&) /ssd_aoem2/sdlcore/sdl_core441/src/components/include/utils/make_shared.h:88
    #2 0x192790f in app_launch::LauncherGenerator::operator()() const /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/app_launch/device_apps_launcher.cc:111
    #3 0x1929c47 in std::back_insert_iterator<std::vector<utils::SharedPtr<app_launch::Launcher>, std::allocator<utils::SharedPtr<app_launch::Launcher> > > > std::generate_n<std::back_insert_iterator<std::vector<utils::SharedPtr<app_launch::Launcher>, std::allocator<utils::SharedPtr<app_launch::Launcher> > > >, unsigned short, app_launch::LauncherGenerator>(std::back_insert_iterator<std::vector<utils::SharedPtr<app_launch::Launcher>, std::allocator<utils::SharedPtr<app_launch::Launcher> > > >, unsigned short, app_launch::LauncherGenerator) /usr/include/c++/5/bits/stl_algo.h:4337
    #4 0x1927c7b in app_launch::DeviceAppsLauncherImpl::DeviceAppsLauncherImpl(app_launch::DeviceAppsLauncher&, app_launch::AppsLauncher&) /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/app_launch/device_apps_launcher.cc:130
    #5 0x1925db9 in app_launch::DeviceAppsLauncher::DeviceAppsLauncher(application_manager::ApplicationManager&, app_launch::AppsLauncher&, app_launch::AppLaunchSettings const&) /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/app_launch/device_apps_launcher.cc:197
    #6 0x191f448 in app_launch::AppLaunchCtrlImpl::AppLaunchCtrlImpl(app_launch::AppLaunchData&, application_manager::ApplicationManager&, app_launch::AppLaunchSettings const&) /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/app_launch/app_launch_ctrl_impl.cc:57
    #7 0x1857b13 in application_manager::ApplicationManagerImpl::Init(resumption::LastState&, media_manager::MediaManager*) /ssd_aoem2/sdlcore/sdl_core441/src/components/application_manager/src/application_manager_impl.cc:2236
    #8 0x17ead91 in main_namespace::LifeCycle::StartComponents() /ssd_aoem2/sdlcore/sdl_core441/src/appMain/life_cycle.cc:116
    #9 0x17fd7c4 in main /ssd_aoem2/sdlcore/sdl_core441/src/appMain/main.cc:137
    #10 0x7f134033082f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)

jacobkeeler, 2020-08-31:
Hello @AlbertGRMN, can you verify if this issue still occurs on the latest release? It looks like similar issues have been submitted before which have been fixed: https://github.com/smartdevicelink/sdl_core/issues/3416

AlbertGRMN, 2020-09-02:
Hi @jacobkeeler , thanks for the reply. The result above is with the patch #2270  and the the leak mentioned in #3416  is gone. However, there's still some other leaks found while streaming and the amount seems to be quite big. It'll increase about 0.3% of memory usage on a 1G RAM machine every hour.

iCollin, 2022-01-18:
Hello @AlbertGRMN, I have found that the destructor for `DeviceAppsLauncherImpl` is not being called, causing both mentioned memory leaks. Both were created at `device_apps_launcher.cc:130` and not destroyed, but by making changes to delete the `DeviceAppsLauncherImpl` when its parent `DeviceAppsLauncher` is destroyed, I no longer see either memory leak.

Do you mind testing the changes in [this commit](https://github.com/smartdevicelink/sdl_core/commit/412f50c7bf9c4fe4119f99e7667989d311028577) and letting me know if they fix the issue for you as well? Thank you!

AlbertGRMN, 2022-01-21:
Hi @iCollin, It looks good.

closed 2022-01-21 by AlbertGRMN