smartdevicelink/sdl_core #3484: SDL sends unexpectedDisconnect=true to HMI in case HMI close an app with RESOURCE_CONSTRAINT reason

dboltovskyi, 2020-08-28:
### Bug Report
SDL sends `unexpectedDisconnect=true` in `BC.OnAppUnregistered` notification to HMI in case HMI close an app using `BC.OnExitApplication` notification with `RESOURCE_CONSTRAINT` reason

##### Occurences: ~5-10%

##### Reproduction Steps
1. SDL and HMI are started
2. App1 and App2 are registered on one mobile connection
3. HMI sends `BC.OnExitApplication` with `RESOURCE_CONSTRAINT` reason related to App2 to SDL

##### Expected Behavior
SDL sends `BC.OnAppUnregistered` notification related to App2 with `unexpectedDisconnect: false`

##### Observed Behavior
SDL sends `BC.OnAppUnregistered` notification related to App2 with `unexpectedDisconnect: true`

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (b476821b44)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - Script: [026_OnExitApplication_RESOURCE_CONSTRAINT_multiple_apps_on_one_connection](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/WebEngine/WebEngineProjection/026_OnExitApplication_RESOURCE_CONSTRAINT_multiple_apps_on_one_connection.lua)
 -

LitvinenkoIra, 2020-09-10:
Luxoft will prepare the fix for this issue

LitvinenkoIra, 2020-09-28:
Please notice that this is a regression issue

jacobkeeler, 2020-09-29:
@LitvinenkoIra how can this be a regression issue if `RESOURCE_CONSTRAINT` was introduced in this release?

dboltovskyi, 2020-09-29:
@jacobkeeler We're apologize for misunderstanding. By saying 'regression' we meant issue needs to be fixed in scope of `7.0` Release since it's related to the feature of this release.

jacobkeeler, 2020-10-23:
Closing with the merge of #3516

closed 2020-10-23 by jacobkeeler