smartdevicelink/sdl_core #3501: SDL sometimes sends Navi.SubscribeVehicleData without parameters during resumption

dboltovskyi, 2020-09-14:
### Bug Report
 SDL sometimes sends Navi.SubscribeVehicleData without parameters during resumption

##### Occurrences:
~1%

##### Reproduction Steps
1. AddSubMenu related to resumption is sent by App1
2. App1 and App2 are subscribed to the same VehicleData
3. Unexpected disconnect and reconnect are performed
4. App1 and App2 re-register with actual HashId

##### Expected Behavior
SDL starts resumption process for App1 and App2
SDL sends `Navi.SubscribeVehicleData` with required parameters

##### Observed Behavior
SDL sends `Navi.SubscribeVehicleData` without parameters

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (a36ea82)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: [038_Another_RPC_VD_RAI2_before_request_to_HMI](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Resumption/Handling_errors_from_HMI/038_Another_RPC_VD_RAI2_before_request_to_HMI.lua)
 - Report and Logs:

dboltovskyi, 2020-09-14:
@theresalech Please notice this is regression issue which is related to implementation of [[SDL 0190] Handle response from HMI during resumption data](https://github.com/smartdevicelink/sdl_core/issues/2486) proposal.
Luxoft team is going to prepare a fix.

theresalech, 2020-09-15:
@dboltovskyi thank you! Please let us know when the fix is ready for Livio review.

PR #3508 (2020-09-22): Fix issue with redundant subscriptions to shared vehicle data during resumption
Fixes #3501

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF test script

### Summary
This PR is intended to fix issue with redundant subscriptions to shared vehicle data during resumption. In several cases there are successful shared subscriptions to vehicle data and also failed other subscriptions. During reverting of all subscriptions for application with failed resumption there is the case when fake response for next waiting request with shared vehic [... cut]
closed 2020-09-22 by JackLivio