smartdevicelink/sdl_core #3502: SDL sometimes sends Navi.SubscribeVehicleData twice during resumption

dboltovskyi, 2020-09-14:
### Bug Report
SDL sometimes sends `Navi.SubscribeVehicleData` twice during resumption

##### Occurrences:
~0.1%

##### Reproduction Steps
1. AddSubMenu related to resumption is sent by App1
2. App1 and App2 are subscribed to the same Vehicle Data
3. Unexpected disconnect and reconnect are performed
4. App1 re-register with actual HashId

##### Expected Behavior
SDL sends `Navi.SubscribeVehicleData` only once

##### Observed Behavior
SDL sends `Navi.SubscribeVehicleData` twice

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (a36ea82)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: [039_Another_RPC_VD_RAI2_before_erroneous_response_from_HMI](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Resumption/Handling_errors_from_HMI/039_Another_RPC_VD_RAI2_before_erroneous_response_from_HMI.lua)
 - Report and Logs:

dboltovskyi, 2020-09-14:
@theresalech Please notice this is regression issue which is related to implementation of [[SDL 0190] Handle response from HMI during resumption data](https://github.com/smartdevicelink/sdl_core/issues/2486) proposal.
Luxoft team is going to prepare a fix.

theresalech, 2020-09-15:
@dboltovskyi thank you! Please let us know when the fix is ready for Livio review.

iCollin, 2020-09-29:
Closed via #3512

PR #3512 (2020-09-29): Add additional check to avoid duplicate subscriptions to Vehicle Data
Fixes #3502

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF test scripts.
Affected unit tests are updated, also new unit test is added.

### Summary
In some situations we need to handle case, when first application has already subscribed to shared data, and second application starts it's resumption by HandleResumptionSubscriptionRequest method. Now check for already existed subscriptions is located in VehicleInfoPlugin, so rarely there is a race condition, when t [... cut]
closed 2020-09-29 by iCollin