smartdevicelink/sdl_core #3503: SDL sometimes doesn't revert the VehicleInfo subscription during resumprion

dboltovskyi, 2020-09-14:
### Bug Report
SDL sometimes doesn't revert the VehicleInfo subscription during resumprion 

##### Occurrences:
~0.3%

##### Reproduction Steps
1. AddCommand, AddSubMenu, CreateInteractionChoiceSet, SetGlobalProperties, SubscribeButton, SubscribeVehicleData,
 SubscribeWayPoints, CreateWindow (<Rpc_n>) are sent by app
2. Unexpected disconnect and reconnect are performed
3. App re-registers with actual HashId
SDL does:
 - start resumption process
 - send set of <Rpc_n> requests to HMI
4. HMI responds with <erroneous> resultCode to one request and <successful> for others
ta

##### Expected Behavior
SDL does:
 - process responses from HMI
 - remove already restored data

##### Observed Behavior
SDL does not remove already restored VehicleInfo data

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (a36ea82)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: [023_resume_all_data_with_one_error_rpc_unexpected_disconnect_limited.lua.tar.gz](https://github.com/smartdevicelink/sdl_core/files/5220807/023_resume_all_data_with_one_error_rpc_unexpected_disconnect_limited.lua.tar.gz)
This is modified version of [023_resume_all_data_with_one_error_rpc_unexpected_disconnect_limited](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Resumption/Handling_errors_from_HMI/023_resume_all_data_with_one_error_rpc_unexpected_disconnect_limited.lua) in order to catch the issue faster
 - Report and Logs:

dboltovskyi, 2020-09-14:
@theresalech Please notice this is regression issue which is related to implementation of [[SDL 0190] Handle response from HMI during resumption data](https://github.com/smartdevicelink/sdl_core/issues/2486) proposal.
Luxoft team is going to prepare a fix.

theresalech, 2020-09-15:
@dboltovskyi thank you! Please let us know when the fix is ready for Livio review.

iCollin, 2020-09-30:
Closed via #3515

PR #3515 (2020-09-30): Unsubscribe from event in VehicleInfoPendingResumptionHandler
Fixes #3503
This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF test scripts

### Summary
Now VehicleInfoPendingResumptionHandler doesn't unsubscribe from already processed event, so there is a case, when VehicleInfoPendingResumptionHandler can catch response with same correlation ID repeatedly. That's why `unsubscribe_from_event()` call is added to method, invoked during processing of corresponding event.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d [... cut]
closed 2020-09-30 by iCollin