smartdevicelink/sdl_core #3511: SDL does not send the value of samplingRate" parameter stored in hmi_capabilities_cache.json in RegisterAppInterface response

VjKlepikov, 2020-09-24:
*Preconditions:*
 # Value of HMICapabilitiesCacheFile parameter is defined (hmi_capabilities_cache.json) in smartDeviceLink.ini file
 # HMI capabilities cache file (hmi_capabilities_cache.json) exists on file system 
 # Value of bitsPerSample is defined *"bitsPerSample" : "16_BIT"* in HMI capabilities cache file (hmi_capabilities_cache.json) 

"audioPassThruCapabilities" : [{ "audioType" : "PCM", *"bitsPerSample" : "16_BIT"*, "samplingRate" : "16KHZ" }],

"pcmStreamCapabilities" : { "audioType" : "PCM", *"bitsPerSample" : "16_BIT"*, "samplingRate" : "16KHZ" }

     4. Value of bitsPerSample is defined *"bitsPerSample" : "RATE_8_BIT"* in default capabilities file (hmi_capabilities.json)

  "audioPassThruCapabilities": [{ "samplingRate": "8KHZ", *"bitsPerSample": "RATE_8_BIT",* "audioType": "PCM" }],
   "pcmStreamCapabilities": { "samplingRate": "8KHZ", *"bitsPerSample": "RATE_8_BIT",* "audioType": "PCM" }

    5. HMI and SDL are started

*Reproduction Steps:*

1. Mobile sends RegisterAppInterface request to SDL

*Observed Behavior:*

SDL sends RegisterAppInterface response with the value of samplingRate" parameter stored in hmi_capabilities_cache.json  to Mobile
RegisterAppInterface:
{ ... "audioPassThruCapabilities" : [\\{ "audioType" : "PCM", *"bitsPerSample" : "8_BIT"*, "samplingRate" : "16KHZ" }],
"pcmStreamCapabilities" : { "audioType" : "PCM", *"bitsPerSample" : "8_BIT"*, "samplingRate" : "16KHZ" }} ... }

 
*Expected Behavior:*

SDL sends RegisterAppInterface response with corresponding capabilities (stored in hmi_capabilities_cache.json) to Mobile
RegisterAppInterface:
{ ... "audioPassThruCapabilities" : [\{ "audioType" : "PCM", *"bitsPerSample" : "16_BIT"*, "samplingRate" : "16KHZ" }],
"pcmStreamCapabilities" : { "audioType" : "PCM", *"bitsPerSample" : "16_BIT"*, "samplingRate" : "16KHZ" }} ... }

AKalinich-Luxoft, 2020-09-24:
@theresalech please note the fix is ready for Livio review

theresalech, 2020-09-24:
@AKalinich-Luxoft thank you! Is this a Core 7.0 regression issue (introduced by a 7.0 feature/bug fix implementation), or does it currently exist on `master`?

AKalinich-Luxoft, 2020-09-24:
@theresalech this issue was found during release 7.0 testing but it is not related to a new features, so I don't think it is a direct regression issue. It is just was not covered by our tests before.

As I can see issue is currently exist on `master` as well. The root cause is that in some places of SDL code there are functions which translate invalid string values into corresponding enum values.
As an example, this line of mapping:
https://github.com/smartdevicelink/sdl_core/blob/0ce76256232bad18bac986e7e68f35c2ae510a29/src/components/application_manager/src/hmi_capabilities_impl.cc#L296

However, `RATE_8_BIT` is not a valid string. It should be `8_BIT` in order to correspond to enum value from "name" attribute:
https://github.com/smartdevicelink/rpc_spec/blob/53e3dfb3cd7ff75a3425312ae4aeb758a1e0fc51/MOBILE_API.xml#L510

Also, we are not sure what was the purpose of "internal_name" attribute but looks like some conversions was based on it as well.

theresalech, 2020-09-24:
@AKalinich-Luxoft, thank you for the additional information! Because this is not a regression issue, and was not included in the Steering Committee-approved issue list for Core 7.0, we will need to place priority on those issues first, but will certainly review this as time allows and make an effort to include in the Core 7.0 release if possible. Thanks!

dboltovskyi, 2020-09-25:
@theresalech Please notice. 
Despite the issue reproducible at previous `sdl_core`releases after additional analysis we came into conclusion it is related to [SDL-0249 Persisting HMI Capabilities](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0249-Persisting-HMI-Capabilities-specific-to-headunit.md) proposal which is in scope of `7.0` release.
Without the fix SDL won't be able to read some values from HMI Capabilities cache.
So we appreciate if PM prioritize this issue as an regression one.

theresalech, 2020-09-28:
@dboltovskyi, thank you for the additional information. We'll do our best to take a look!

jacobkeeler, 2020-09-29:
Closing with the merge of #3510

PR #3510 (2020-09-29): Fix/fix enums conversion in hmi capabilities
Fixes #3511 

This PR is **not ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF and manual testing

### Summary
There were found a couple of hardcoded HMI capabity enum to string values and some of them were outdated. This legacy
code can be easily replaced with the templated function supplied from generated API. By that reason, all hardcoded pairs and maps were removed and replaced with a common function.

### Tasks Remaining:
- [x] Double check unit tests

### CLA
- [x] I have signed [the CLA](https: [... cut]
closed 2020-09-29 by jacobkeeler