smartdevicelink/sdl_core #3514: SDL sends GetInteriorVehicleData response with moduleData = null in case of erroneous response type for GetInteriorVehicleData from HMI

AKalinich-Luxoft, 2020-09-24:
**Preconditions:**
1) sdl_preloaded_pt.json default app_policies
groups: [ "Base-4", "RemoteControl" ]
"moduleType": [ "RADIO", "CLIMATE", "SEAT", "AUDIO", "LIGHT", "HMI_SETTINGS" ]
HMI supports RC functionality
2) SDL and HMI started

**Reproduction Steps:**
1. App→SDL: GetInteriorVehicleData("moduleType": "module_1", subscribe = true )
SDL→HMI: RC.GetInteriorVehicleData("moduleType": "module_1", subscribe = true )
HMI→SDL: RC.GetInteriorVehicleData( GENERIC_ERROR)
{"error":{"code":22,"message":"Erroneous response is assigned by settings","data": {"method":"RC.GetInteriorVehicleData"}
}}

**Observed Behavior:**
SDL→App: GetInteriorVehicleData("moduleData":null, "resultCode": "GENERIC_ERROR", "success": "false", info )

**Expected Behavior:**
SDL→App: GetInteriorVehicleData( "resultCode": "GENERIC_ERROR", "success": "false", info )

**OS & Version Information**
https://github.com/LuxoftSDL/sdl_core/tree/develop (a36ea82) 
https://github.com/LuxoftSDL/sdl_hmi/tree/develop (6d32f1b)

dboltovskyi, 2020-09-30:
@theresalech Please notice this is not regression issue and it's applicable for Release `6.1.1`

theresalech, 2020-09-30:
@dboltovskyi, thank you for letting us know. We are currently focused on regression issues (issues introduced during Core 7.0 development), and existing issues that were previously prioritized by the Steering Committee for the Core 7.0 release. We'll therefore look at this issue/PR when time allows, following the regression and previously prioritized issues. Thanks!

AKalinich-Luxoft, 2020-11-05:
@theresalech Please notice PR is ready for PM review.

theresalech, 2020-11-09:
@AKalinich-Luxoft, thank you for letting us know! This issue was not requested as priority for the next Core release, but we will try to review as time allows. If it's imperative this be included in the next release, please raise as priority to the SDLC Steering Committee. Thanks!

ShobhitAd, 2021-03-05:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3565

PR #3565 (2021-03-05): Fix null moduleData in GetIVD response
Fixes #3514 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by manual testing

### Summary
Removed `msg_params` section forwarding for case when GetIVD response was not successful

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2021-03-05 by ShobhitAd