smartdevicelink/sdl_core #3517: SDL sends GetInteriorVehicleData response with "success":false in case success type of response(resultCode: WRONG_LANGUAGE, RETRY, SAVED) from HMI

VjKlepikov, 2020-09-25:
*Preconditions:*
1. SDL and HMI started

*Reproduction Steps:*
# Mobile app sends GetInteriorVehicleData("module_1",  subscribe = true) request
*App→SDL:* GetInteriorVehicleData("moduleType": "module_1",  subscribe = true )

*Observed Behavior:*
*SDL→HMI:* RC.GetInteriorVehicleData("moduleType": "module_1", subscribe = true )
*HMI→SDL:* RC.GetInteriorVehicleData(*"code":16*, "moduleType": "module_1",* isSubscribed = true* , values same as shown on HMI, SUCCESS)
*SDL→App:* GetInteriorVehicleData(*"success":false*, "moduleType": "module_1", values same as shown on HMI, SUCCESS)

*Expected Behavior:*

*SDL→HMI:* RC.GetInteriorVehicleData("moduleType": "module_1", subscribe = true )
*HMI→SDL:* RC.GetInteriorVehicleData(*"code":16*, "moduleType": "module_1", isSubscribed = true , values same as shown on HMI, SUCCESS)
*SDL→App:* GetInteriorVehicleData(*"success":true*, "moduleType": "module_1", *isSubscribed = true* , values same as shown on HMI, SUCCESS)
*SDL→App:* OnHashChange(hashID) notification

dboltovskyi, 2020-09-25:
@theresalech Please notice this issue related to the [SDL-0188 Interior Vehicle Data resumption](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0188-get-interior-data-resumption.md) proposal which is in scope of `7.0` Release. It aligns error handling for `GetInteriorVehicleData` RPC in general and resumption scenarios.
Luxoft team is going to prepare a fix.

theresalech, 2020-09-28:
@dboltovskyi, please advise when a fix is ready for Livio's review. Thank you!

iCollin, 2020-10-05:
Closed via #3519

PR #3519 (2020-10-02): Fix GetInteriorVehicle request wrong processes result codes from HMI
Fixes #3517 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF script : https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2460
Unit test

### Summary
GetInteriorVehicle request RPC should process the next result codes as success: WRONG_LANGUAGE, RETRY, SAVED.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-10-05 by iCollin