smartdevicelink/sdl_core #3518: SDL sometimes sends Navi.SubscribeWayPoints twice during resumption

dboltovskyi, 2020-09-25:
### Bug Report
SDL sometimes sends `Navi.SubscribeWayPoints` twice during resumption

##### Occurrences:
~0.3%

##### Reproduction Steps
1. AddSubMenu related to resumption is sent by App1
2. App1 and App2 are subscribed to Way Points
3. Unexpected disconnect and reconnect are performed
4. App1 re-register with actual HashId

##### Expected Behavior
SDL sends `Navi.SubscribeWayPoints` only once

##### Observed Behavior
SDL sends `Navi.SubscribeWayPoints` twice

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (a36ea82)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: [041_Another_RPC_WP_RAI2_before_request_to_HMI](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Resumption/Handling_errors_from_HMI/041_Another_RPC_WP_RAI2_before_request_to_HMI.lua)
 - Report and Logs:

dboltovskyi, 2020-09-25:
@theresalech Please notice this is regression issue which is related to implementation of [[SDL 0190] Handle response from HMI during resumption data](https://github.com/smartdevicelink/sdl_core/issues/2486) proposal.
Luxoft team is going to prepare a fix.

iCollin, 2020-10-19:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3531

PR #3531 (2020-10-19): Fix synchronization in WaypointsPendingResumptionHandler
Fixes #3518

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
In some situations we need to handle case, when first application has already subscribed to shared data, and second application starts it's resumption by HandleResumptionSubscriptionRequest method. Now check for already existed subscriptions is located in SDLRPCPlugin, so rarely there is a race condition, when this check is performed for second application too early right before first application has completed it's subscription process. [... cut]
closed 2020-10-19 by iCollin