smartdevicelink/sdl_core #3522: SDL sometimes finishes resumption successfully even if error has been received from HMI

dboltovskyi, 2020-09-30:
### Bug Report
SDL sometimes finishes resumption successfully even if error has been received from HMI 

##### Occurrences:
~0.01% (very rare)

##### Reproduction Steps
1. App sends `AddCommand` request (which is stored by SDL as resumption data)
2. App adds some other resumption data (e.g. SubMenu, VD and WP subscription etc.)
3. Unexpected disconnect and reconnect are performed
4. App re-register with actual HashId
5. SDL starts resumption process for App and sends all required requests to HMI
6. HMI replies with erroneous response to one of the requests (e.g. `VR.AddCommand`)

##### Expected Behavior
SDL fails resumption and send `RESUME_FAILED` to app

##### Observed Behavior
SDL successfully finishes resumption and send `SUCCESS` to app

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (a36ea82)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: [004_resume_one_rpc_with_error_result_code_ign_off](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Resumption/Handling_errors_from_HMI/004_resume_one_rpc_with_error_result_code_ign_off.lua)
 - Report and Logs:

dboltovskyi, 2020-10-01:
@theresalech Please notice this issue is related to [SDL-0190 Handle response from HMI during resumption data](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0190-resumption-data-error-handling.md) proposal which is in scope of `7.0` Release

theresalech, 2020-10-01:
@dboltovskyi, thank you. Please let us know once the fix is ready for Livio's review.

iCollin, 2020-10-05:
Closed via #3525

PR #3525 (2020-10-05): Add check for already processed resumptions to ResumptionDataProcessor
Fixes #3522

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
This PR is intended to improve synchronization in `ResumptionDataProcessorImpl::Restore()` method. Root cause of this problem is a case when some requests from HMI may be processed earlier than callback will be set in ResumptionDataProcessor. So in this case execution of `ProcessResumptionStatus` method for such application will be aborted due to absence of callback, and this behavior is valid and expected. But after this  [... cut]
closed 2020-10-05 by iCollin