smartdevicelink/sdl_core #3527: SDL sometimes deactivates app to BACKGROUND after successful HMI level resumption

dboltovskyi, 2020-10-02:
### Bug Report
SDL sometimes deactivates app to BACKGROUND after HMI level resumption

##### Occurrences:
~3%

##### Reproduction Steps
1. There are 2 applications registered: `App_1` and `App_2`
`App_1` in `LIMITED`, `App_2` in `FULL` HMI level
2. Both apps are subscribed to some module of Interior Vehicle Data
3. Unexpected disconnect and reconnect are performed
4. Both apps re-register with actual HashId
5. SDL successfully finished data resumption
=> Restored subscription to Interior Vehicle Data for both apps
6. SDL successfully finished HMI level resumption:
=> `App_1` restored to `LIMITED`, `App_2` restored to `FULL`

##### Expected Behavior
SDL doesn't change HMI level of applications

##### Observed Behavior
SDL change HMI level of `App_2` from `FULL` to `BACKGROUND` by sending additional not required `OnHMIStatus` notification

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (a36ea82)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Scripts:
    - [015_InteriorVD_resumption_two_apps_same_moduleTypes_same_moduleIds_unexpected_disconnect](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Resumption/InteriorVehicleData/015_InteriorVD_resumption_two_apps_same_moduleTypes_same_moduleIds_unexpected_disconnect.lua)
   - [016_InteriorVD_resumption_two_apps_same_moduleTypes_same_moduleIds_IGN_OFF](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Resumption/InteriorVehicleData/016_InteriorVD_resumption_two_apps_same_moduleTypes_same_moduleIds_IGN_OFF.lua) 
 - Report and Logs:

dboltovskyi, 2020-10-05:
@theresalech Please notice we are not 100% sure whether this is regression issue or not.
ATF scripts which uncover it have been created in scope of `0188` feature. However new functionality implemented in scope of this feature is not related to HMI levels. We also haven't seen such issue before in previous releases when running other ATF scripts.
We appreciate if PMs are able to analyze this issue if time allows.

JackLivio, 2020-10-06:
@dboltovskyi I looked into this issue and heres what i found.

App2 is being set to background because of this statement: https://github.com/smartdevicelink/sdl_core/blob/release/7.0.0/src/components/application_manager/src/state_controller_impl.cc#L411

In most cases where this test passes, that same resolver function usually exits here because app2 still has HMILevel:None before it was assigned FULL like in the failure case: https://github.com/smartdevicelink/sdl_core/blob/release/7.0.0/src/components/application_manager/src/state_controller_impl.cc#L375

I think the logic in this block is flawed: https://github.com/smartdevicelink/sdl_core/blob/release/7.0.0/src/components/application_manager/src/state_controller_impl.cc#L400-L412

Even though app2 is not streamable and is not audible, it should still be able to resume to full, right?

JackLivio, 2020-10-06:
This fix seemed to resolve the issue. Might be too specific of a case? Not sure if there are other cases that would be affected.

```
  if (mobile_apis::VideoStreamingState::STREAMABLE == result_video_state ||
      Compare<mobile_apis::AudioStreamingState::eType, EQ, ONE>(
          result_audio_state,
          mobile_apis::AudioStreamingState::AUDIBLE,
          mobile_apis::AudioStreamingState::ATTENUATED)) {
    result_hmi_level =
        mobile_apis::PredefinedWindows::DEFAULT_WINDOW == window_id_ &&
                applied_->IsFullscreen()
            ? mobile_apis::HMILevel::HMI_LIMITED
            : to_resolve_hmi_level;
    SDL_LOG_DEBUG("if result hmi level: " << result_hmi_level);
  } else if (mobile_apis::HMILevel::HMI_FULL == to_resolve_hmi_level &&
             mobile_apis::HMILevel::HMI_LIMITED == applied_hmi_level) {
    result_hmi_level = mobile_apis::HMILevel::HMI_FULL;
  } else {
    SDL_LOG_DEBUG("Else setting to hmi_background");
    result_hmi_level = mobile_apis::HMILevel::HMI_BACKGROUND;
  }
```

dboltovskyi, 2020-10-06:
@JackLivio Thank you for your analysis.
> Even though app2 is not streamable and is not audible, it should still be able to resume to full, right?

Yes, that is correct.

Unfortunately we can't comment on the change suggested from developer perspective due to lack of resources.
However we can propose to verify this change by running existing Unit Tests and ATF tests and check if everything is Ok.

I'll post the results once I get them.

dboltovskyi, 2020-10-06:
@JackLivio Please notice all UT and ATF tests are passed using suggested solution.

aderiabin, 2020-10-09:
Steps to reproduce the issue (100% occurrence):
 1. Start WebHMI and SDL
 2. Register 2 mobile applications (App1 - media, App2 - not media)
 3.Pass applications into next HMI levels: App1 - LIMITED, App2 - FULL
 4. Send valid `SubtleAlert` RPC with TTS from App1

dboltovskyi, 2020-10-13:
@theresalech Please notice fix in `sdl_core` and update in `sdl_atf_test_scripts` are ready for PM review.

dboltovskyi, 2020-10-13:
We would prioritize this issue for `7.0` release since it can be easily reproduced within new `SubtleAlert` RPC which has been introduced in this release.

PR #3538 (2020-10-20): Keep FULL for apps after conflict resolution
Fixes #3527 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF script

### Summary
There was found an issue in state controller related to wrong transition of non-media applications (not streamable and not audible) from FULL to BACKGROUND in case of some events. To fix that issue, SDL should check first if non-media application is currently in FULL and another application is not in FULL. If so, non-media application should stay in FULL.

### CLA
- [x] I have signed [the CLA](https://docs.g [... cut]
closed 2020-10-20 by JackLivio