smartdevicelink/sdl_core #3532: SDL does not transfer OnSystemCapabilityUpdated notification to App

VjKlepikov, 2020-10-07:
### Bug Report
SDL does not transfer OnSystemCapabilityUpdated notification from HMI to an App 

#### Preconditions:
1) In sdl_preloaded_pt.json default app_policies groups:
["Base-4", "WidgetSupport" ]
2) SDL and HMI are started
3) App is registered and activated(App has hmi level FULL/LIMITED )
4) App created and activated a widget ( the widget has "hmi_level": FULL )
5) IGN_OFF and IGN_ON are performed
6) App is registered with actual hashId
7) Widget is activated

##### Reproduction Steps
App sends "Show" request with "templateConfiguration" param for the restored widget
HMI sends BC.OnSystemCapabilityUpdated:
HMI→SDL: "BC.OnSystemCapabilityUpdated"(params)

##### Expected Behavior
SDL→App: “OnSystemCapabilityUpdated” (params)

##### Observed Behavior
SDL does not transfer OnSystemCapabilityUpdated(params) to Mobile App

##### OS & Version Information
SDL: https://github.com/smartdevicelink/sdl_core/commits/develop (9f61819)
HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop (f712d4e)

##### Test Case, Sample Code, and / or Example App

VjKlepikov, 2020-10-16:
@theresalech Please notice this is regression issue

iCollin, 2020-10-22:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3550

closed 2020-10-22 by iCollin