smartdevicelink/sdl_core #3535: SDL core dump during app activation

dboltovskyi, 2020-10-08:
### Bug Report
SDL core dump observed during app activation. 
Issue is reproduced only at `Ubuntu 16.04` and not reproducible for `Ubuntu 18.04`

##### Reproduction Steps
1. Start SDL, init HMI, connect Mobile device
2. Register new mobile Application
3. Activate application on HMI

##### Expected Behavior
Application activated successfully

##### Observed Behavior
SDL crashed

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/7.0.0 (87ff1bb)
* Testing Against: SPT

##### Test Case, Sample Code, and / or Example App
 - Trace: [Trace.txt.tar.gz](https://github.com/smartdevicelink/sdl_core/files/5349188/Trace.txt.tar.gz)
 - SDL log: [SmartDeviceLinkCore.log.tar.gz](https://github.com/smartdevicelink/sdl_core/files/5349185/SmartDeviceLinkCore.log.tar.gz)

dboltovskyi, 2020-10-08:
Issue is not reproduced at  https://github.com/smartdevicelink/sdl_core/tree/release/7.0.0 (876856f)

dboltovskyi, 2020-10-08:
@theresalech Please notice this is regression issue

iCollin, 2020-10-08:
Thank you for noticing. Closed via #3536

PR #3536 (2020-10-08): remove unlock of lock that isn't locked
Fixes #3535

Since #3497 this lock will not be locked ever when this unlock line is executed because the autolock is destroyed when `LoadPolicyManager` returns.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-10-08 by iCollin