smartdevicelink/sdl_core #3539: SDL does not send reason info in NACK message to mobile app in specific cases

GetmanetsIrina, 2020-10-13:
### Bug Report
SDL does not send reason info in NACK messages to a mobile app in specific cases

##### Reproduction Steps
1. SDL and HMI are started
2. Mobile app is registered via 5.3 protocol and with `NAVIGATION` HMI type

##### Reproduction Steps
1. Mobile app requests start service in not appropriate HMI level.
or
2. Mobile app requests start service with not appropriate HMI type (not NAVI)
or
3. Mobile app requests protected start service in case service should be not protected according to settings in *.ini file
or
4. Mobile app requests unprotected start service in case service should be protected according to settings in *.ini file
or
5. Mobile app requests start of unprotected service in case it has already started

##### Expected Behavior
SDL provides a reason in NACK message to mobile app

##### Observed Behavior
SDL does not provide a reason in NACK message to mobile app

##### OS & Version Information
* OS/Version: 18.04
* SDL Core Version: release/7.0.0 (fd291d76)
* HMI: develop(6d32f1ba)
* ATF: develop

GetmanetsIrina, 2020-10-13:
@theresalech Please notice the issue is related to [SDL-0308 Add a Reason Parameter to All Protocol NAKs](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0308-protocol-nak-reason.md) proposal which is in the scope of 7.0 Release.
Luxoft team is going to prepare a fix.

theresalech, 2020-10-13:
@GetmanetsIrina, thanks for letting me know. Please advise when Luxoft has a PR ready for Livio review!

jacobkeeler, 2020-10-23:
Closed via #3545

PR #3545 (2020-10-23): Add missed NACK reasons
Fixes #3539 

This PR is **ready** for review.

### Risk
This PR makes **major** API changes.

### Testing Plan
Covered by ATF test scripts
Affected UTs are updated

### Summary
In several cases there is a need in adding or rewording reasons of NACK, related to Video/Audio services. To fix cases with wrong application types and wrong HMI Levels there is a need in adding new reason parameter to two methods: ConnectionHandlerImpl::NotifyServiceStartedResult and ApplicationManagerImpl::OnStreamingConfigurationFailed. Last one is a result of refactoring of ApplicationManage [... cut]
closed 2020-10-23 by jacobkeeler