smartdevicelink/sdl_core #3542: SDL is unable to activate application after resumption in case of two apps

dboltovskyi, 2020-10-16:
### Bug Report
SDL is unable to activate application after resumption in case of two apps

##### Reproduction Steps
1. There are two `MEDIA` apps registered: `App_1` and `App_2`
2. `App_1` is in `LIMITED` and `App_2` is in `BACKGROUND` HMI levels
3. Unexpected disconnect and reconnect are performed
4. SDL resumed HMI level of both apps:
=> `App_1` is in `LIMITED`, `App_2` is in `BACKGROUND` HMI levels
5. `App_2` activated
6. `App_1` activated

##### Expected Behavior
SDL does:
  - send `OnHMIStatus(FULL)` for `App_1`
  - send `OnHMIStatus(BACKGROUND)` for `App_2`

##### Observed Behavior
SDL does:
 - reply to `SDL.ActivateApp` request from HMI
 - send `OnHMIStatus(NONE)` for `App_1`
 - not send `OnHMIStatus` for `App_2`

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/7.0.0 (5c72cddc35)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script:
 - Report and Logs:

dboltovskyi, 2020-10-16:
Please notice issue is applicable for [6.1.0](https://github.com/smartdevicelink/sdl_core/tree/release/6.1.0) and [6.0.1](https://github.com/smartdevicelink/sdl_core/tree/release/6.0.1) releases, however is not reproduced at [5.1.3](https://github.com/smartdevicelink/sdl_core/tree/5.1.3)
Probably the root cause is the fix https://github.com/smartdevicelink/sdl_core/pull/3117 which was included in `6.0.1`

dboltovskyi, 2020-10-16:
@theresalech We would like to treat this issue as regression since it's not reproduced at `5.1.3` and steps to reproduce are quite simple.

theresalech, 2020-10-19:
@dboltovskyi will Luxoft be submitting a PR to resolve this issue?

dboltovskyi, 2020-10-19:
@theresalech Yes, Luxoft team is going to prepare a fix

theresalech, 2020-10-19:
@dboltovskyi, thanks for the quick response! Please let us know once the PR is ready for review. If this issue existed in previous versions of SDL Core, we will not be considering it a regression issue (meaning, introduced in Core 7.0), but will work to review and include in the Core 7.0 release if time allows.

iCollin, 2020-10-22:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3550

PR #3550 (2020-10-22): Fix application activation after resumption
Fixes #3542 #3532 
This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF test scripts

### Summary
There is a case when media application acquires wrong HMI Level (NONE instead of FULL) during activation. Root cause of this issue is a `true` value of `is_resuming` flag for application, which in fact completed resumption process. Before this flag was set to false only for applications with failed resumption, so there is a need to assign it to false for all already resumed applications regardless  [... cut]
closed 2020-10-22 by iCollin