smartdevicelink/sdl_core #3547: SDL sometimes doesn't send required number of StartStream/StartAudioStream requests

dboltovskyi, 2020-10-21:
### Bug Report
During start of Video(Audio) streaming SDL sends `Navigation.StartStream` (`Navigation.StartAudioStream`) requests to HMI and waiting for response. In case if HMI responds with erroneous code SDL re-send the same request several times (equal to `StartStreamRetry` .ini parameter). SDL sometimes doesn't send required number of such requests

##### Occurrences
~10-15%

##### Reproduction Steps
1. Navi app is registered and activated
2. App try to start Video(Audio) service
=> SDL does send `Navi.StartStream` (`Navi.StartAudioStream`) requests to HMI
3. HMI responds with `REJECTED`

##### Expected Behavior
SDL re-sends `Navi.StartStream` (`Navi.StartAudioStream`) requests to HMI a few times (equal to `StartStreamRetry` .ini parameter)

##### Observed Behavior
SDL doesn't send required number of requests
There is an error in log for Video:
```
ERROR [21 Oct 2020 15:56:56,257][140043616237312][Commands] /home/developer/sdl/sdl_core/src/components/application_manager/rpc_plugins/sdl_rpc_plugin/src/commands/hmi/navi_start_stream_request.cc:155 commands::NaviStartStreamRequest::RetryStartSession: NaviStartStreamRequest aborted. Application not found
```
There is an error in log for Audio:
```
ERROR [21 Oct 2020 16:29:37,566][0x7f311ffff700][Commands] /home/developer/sdl/sdl_core/src/components/application_manager/rpc_plugins/sdl_rpc_plugin/src/commands/hmi/navi_audio_start_stream_request.cc:156 commands::AudioStartStreamRequest::RetryStartSession: StartAudioStreamRequest aborted. Application not found
```

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/7.0.0 (3ec2b6b)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: 
 - Report and Logs:

dboltovskyi, 2020-10-21:
Issue is also applicable for `6.1.1` release

dboltovskyi, 2020-10-21:
@theresalech Please notice this is regression issue. Luxoft team is going to submit a fix.

theresalech, 2020-10-21:
@dboltovskyi, we are defining regression issues as issues introduced by Core 7.0 development. If this issue is present in 6.1.1, it was not introduced in 7.0 and is therefore not considered regression. However, if you are able to submit a fix, we will do our best to review and include in the next 7.0 Release Candidate, which will be delivered to the Steering Committee for approval on November 2. Note that issues introduced by Core 7.0 will take priority at this time. Thank you!

dboltovskyi, 2020-10-23:
@theresalech Please notice we found out this issue may lead to core dump in some scenarios (please see setails in https://github.com/smartdevicelink/sdl_core/pull/3552)
We would appreciate for PM review of the fix in `sdl_core` and update in `sdl_atf_test_scripts` if time allows.

iCollin, 2020-10-26:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3552

PR #3552 (2020-10-26): Fix streaming retry sequence
Fixes #3547 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF scripts

### Summary
There were noticed two issues related to a/v streaming:
1. Due to object self destruction after `TerminateRequest()` call any attempt to access member field of that object may cause an
undefined behavior - it might be a core crash or corrupted value sometimes. In this particular case, SDL tries to access `message_` field through `application_id()` function after object destruction. As a result, sometimes SDL crashes and so [... cut]
closed 2020-10-26 by iCollin