smartdevicelink/sdl_core #3549: SDL sometimes sends OnHMIStatus in wrong order when app is activated too fast by HMI

dboltovskyi, 2020-10-21:
### Bug Report
SDL sometimes sends OnHMIStatus in wrong order when app is activated too fast by HMI.
It's very unlikely issue may be faced with mobile apps, however it may happen with web apps.

##### Occurrences
~20-50%

##### Reproduction Steps
1. App is registered
2. HMI activates app just after receiving `BC.OnAppRegistered` notification

##### Expected Behavior
SDL sends 2 `OnHMIStatus` notifications to App: `NONE`, `FULL`

##### Observed Behavior
SDL sends 2 `OnHMIStatus` notifications to App: `FULL`, `NONE`

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/7.0.0 (3ec2b6b)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: 
 - Report and Logs:

dboltovskyi, 2020-10-21:
@theresalech Please notice this issue is not reproducible at `6.1.1` release, so it can be treated as regression one. Luxoft team is going to provide a fix.

iCollin, 2020-10-30:
Closed via #3559

closed 2020-10-30 by iCollin