smartdevicelink/sdl_core #3556: SDL does not respond to SDL.GetListOfPermission with appId in case there are no specific permissions for the app

GetmanetsIrina, 2020-10-29:
### Bug Report
SDL does not respond to SDL.GetListOfPermission with appId in case there are no specific permissions for the app

##### Reproduction Steps
1. SDL with external_proprietary policies and HMI are started
2. Mobile app is registered and activated
3. There are no specific permissions for the app
4. PTU is not performed or performed without specific permissions for the mobile app
5. Go to HMI settings-> App permissions - > <app name>
6. HMI requests `{"jsonrpc":"2.0","id":<n>,"method":"SDL.GetListOfPermissions","params":{"appID":<app_id>}}`

##### Expected Behavior
SDL responds with {"id":<n>,"jsonrpc":"2.0","result":{"allowedFunctions":[],"code":0,"externalConsentStatus":[],"method":"SDL.GetListOfPermissions"}}

##### Observed Behavior
SDL does not respond to `SDL.GetListOfPermissions` request

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/54246454776513b68161258b46299b48e4352f3f
* Testing Against: HMI: https://github.com/smartdevicelink/sdl_hmi/commit/931877cb68e56522d9f6db1d6c5751decf6c988e, SPT: 20201007

##### SDL log
[SDL_log_no_response_to_GetListOfPermissions.zip](https://github.com/smartdevicelink/sdl_core/files/5457962/SDL_log_no_response_to_GetListOfPermissions.zip)

LitvinenkoIra, 2021-02-08:
@VadymLuchko is working on it

jacobkeeler, 2021-04-22:
Closing with the merge of #3627

PR #3627 (2021-04-22): fix respond on SDL.GetListOfPermission request
Fixes #3556

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by unit test

### Summary
There was noticed an incorrect behavior in PolicyHandler::OnGetListOfPermissions method.
No response was sent if the list of permission was empty.
In this fix we check if we went through all the methods for filling the List Of Permissions and didn't cause an error, then we send an empty list.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6 [... cut]
closed 2021-04-22 by jacobkeeler