smartdevicelink/sdl_core #3561: SDL sometimes doesn't send BC.OnAppUnregistered in case HMI close an app with RESOURCE_CONSTRAINT reason

dboltovskyi, 2020-10-30:
### Bug Report
SDL sometimes doesn't send BC.OnAppUnregistered in case HMI close an app with RESOURCE_CONSTRAINT reason.

##### Occurrences
~0.5%

##### Reproduction Steps
1. SDL and HMI are started
2. App is registered on default mobile connection
3. App is activated
4. HMI sends `BC.OnExitApplication` with reason: `RESOURCE_CONSTRAINT` related to App to SDL

##### Expected Behavior
SDL sends `BC.OnAppUnregistered` notification to HMI

##### Observed Behavior
SDL doesn't send notification to HMI

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/7.0.0 (3ec2b6b)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: [025_OnExitApplication_RESOURCE_CONSTRAINT_one_app](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/WebEngine/WebEngineProjection/025_OnExitApplication_RESOURCE_CONSTRAINT_one_app.lua) - needs to me run 1000 times
 - Report and Logs:

AKalinich-Luxoft, 2020-11-02:
@theresalech please notice this is not a regression issue related to wrong behavior of TCP transport with a pretty low occurrence.
PR with fix is linked and ready for Livio review

theresalech, 2020-11-02:
@AKalinich-Luxoft, thank you for letting us know! If Luxoft would like to make this issue a priority for the next Core release (planned for April 2021), please make sure it is raised to the SDLC Steering Committee as such.

ShobhitAd, 2021-11-09:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3566

closed 2021-11-09 by ShobhitAd