smartdevicelink/sdl_core #3564: SDL sometimes sends 2 requests to HMI with the same id

dboltovskyi, 2020-11-05:
### Bug Report
SDL sometimes sends 2 requests to HMI with the same id. 
There is a requirement for SDL that each request should have unique Id.

##### Reproduction Steps
1. Start SDL
2. Start HMI
3. Check Ids of requests from SDL to HMI

##### Expected Behavior
Ids are unique

##### Observed Behavior
Sometimes Ids may be the same for 2 different requests:
```
SDL->HMI [13-10-2020 17:04:39,433] {"id":1,"jsonrpc":"2.0","method":"BasicCommunication.UpdateDeviceList"}
SDL->HMI [13-10-2020 17:04:39,438] {"id":1,"jsonrpc":"2.0","method":"BasicCommunication.GetSystemInfo"}
```

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/7.0.0 (80973f4)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: Any
 - Report and Logs:

AKalinich-Luxoft, 2020-11-09:
@JackLivio @theresalech please notice this PR is ready for Livio review

theresalech, 2020-11-09:
@AKalinich-Luxoft thank you for letting us know! This issue was not requested as priority for the next Core release, but we will try to review as time allows. If it's imperative this be included in the next release, please raise as priority to the SDLC Steering Committee. Thanks!

jacobkeeler, 2021-02-04:
closing with the merge of #3567

PR #3567 (2021-02-04): Make correlation ID counter atomic
Fixes #3564 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
There is a situation when messages are simultaneously created for different requests. During message creation process `GetNextHMICorrelationID`/`GetNextMobileCorrelationID` methods are invoked, and inside this methods corresponding counters are incremented. But now these counters aren't synchronized properly, because several threads may increment these counters at the same time, so `GetNextHMICorrelationID`/`GetNextMobileCorrelationID` methods may return sa [... cut]
closed 2021-02-04 by jacobkeeler