smartdevicelink/sdl_core #3576: SDL allows starting Video service for 2nd version of SDL protocol

dboltovskyi, 2020-11-17:
### Bug Report
SDL allows starting Auido/Video service for 2nd version of SDL protocol

##### Reproduction Steps
1. Register Navi app with 2nd version of SDL protocol
2. Start Video service

##### Expected Behavior
SDL responds with `StartServiceNACK`

##### Observed Behavior
SDL responds with `StartServiceACK`

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (7871541)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: [1912_Video_service_start_via_2_protocol](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1912_Video_service_start_via_2_protocol.lua)
 - Report and Logs:

dboltovskyi, 2020-11-17:
Issue is not reproduced at [7.0.0](https://github.com/smartdevicelink/sdl_core/tree/7.0.0) and seems introduced by https://github.com/smartdevicelink/sdl_core/pull/3572

dboltovskyi, 2020-11-17:
Scripts affected:
```
./test_scripts/Defects/4_5/1912_Video_service_start_via_2_protocol.lua
./test_scripts/MobileProjection/Phase1/009_Rejecting_audio_video_service_in_FULL_2_protocol.lua
./test_scripts/MobileProjection/Phase1/010_Rejecting_audio_video_service_in_LIMITED_2_protocol.lua
./test_scripts/TheSameApp/Protocol/005_Protocol_StartService_from_2_apps_with_same_appName_diff_appID_DIFF_devices.lua
./test_scripts/TheSameApp/Protocol/006_Protocol_StartService_from_2_apps_with_same_appName_same_appID_DIFF_devices.lua
```

jacobkeeler, 2020-12-15:
Closing with the merge of #3600

PR #3600 (2020-12-15): Bind protocol version when app registers
Fixes #3576 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Run ATF tests mentioned in issue

### Summary
Binds a protocol version to a session when an app registers. Core does call to bind earlier but Core binds the module max version instead of the negotiated version which is decided by the app.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-12-15 by jacobkeeler