smartdevicelink/sdl_core #3577: SDL doesn't allow VD parameters if parameters section is omited in policies for OnVehicleData

dboltovskyi, 2020-11-17:
### Bug Report
SDL doesn't allow VD parameters if parameters section is omited in policies for `OnVehicleData`

In general there are 3 options related to parameters for RPC in policies:
  1. `parameters` section is omitted - allow all parameters
  2. `parameters` section contains empty array (e.g. `[]`) - disallow all parameters
  3. `parameters` section contains some parameters (e.g. `["speed, "gps"]`) - allow only defined parameters

##### Reproduction Steps
1. Make sure there is a functional group in policies with `OnVehicleData` but with omitted `parameters` section
1. Register app
2. Subscribe for some Vehicle Data
3. Send `VI.OnVehicleData` from HMI with any VD parameter

##### Expected Behavior
SDL allows transferring `OnVehicleData` to App

##### Observed Behavior
SDL doesn't allow transferring `OnVehicleData` to App

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (9dc3c070c5)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: [006_Two_apps_with_same_appName_DIFF_devices_send_subscribe_on_VehicleData](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/TheSameApp/Subscriptions/006_Two_apps_with_same_appName_DIFF_devices_send_subscribe_on_VehicleData.lua)
 - Report and Logs:

dboltovskyi, 2020-11-17:
Issue is not reproduced at [7.0.0](https://github.com/smartdevicelink/sdl_core/tree/7.0.0) and seems introduced by https://github.com/smartdevicelink/sdl_core/pull/3568

dboltovskyi, 2020-11-17:
Scripts affected:
```
./test_scripts/TheSameApp/Subscriptions/006_Two_apps_with_same_appName_DIFF_devices_send_subscribe_on_VehicleData.lua
./test_scripts/TheSameApp/Subscriptions/007_Two_apps_with_same_appName_DIFF_devices_send_subscribe_unsubscribe_on_VehicleData.lua
./test_scripts/Resumption/Handling_errors_from_HMI/001_successful_resumption_SUCCESS_resultCode.lua
./test_scripts/Resumption/Handling_errors_from_HMI/002_successful_resumption_other_successful_result_codes.lua
./test_scripts/Resumption/Handling_errors_from_HMI/005_resume_all_data_with_one_error_rpc_unexpected_disconnect.lua
./test_scripts/Resumption/Handling_errors_from_HMI/006_resume_all_data_with_one_error_rpc_ign_off.lua
./test_scripts/Resumption/Handling_errors_from_HMI/009_restore_subscription_for_app2_after_fail_for_app1.lua
./test_scripts/Resumption/Handling_errors_from_HMI/010_restore_subscription_for_app2_after_fail_in_sub_resultCode_for_app1.lua
./test_scripts/Resumption/Handling_errors_from_HMI/011_unsubscribe_from_the_same_vehicledata_without_request_to_hmi_by_error_during_resumption.lua
./test_scripts/Resumption/Handling_errors_from_HMI/012_unsubscribe_from_the_same_vehicledata_without_request_to_hmi_by_error_in_sub_resultCode_during_resumption.lua
./test_scripts/Resumption/Handling_errors_from_HMI/032_Same_RPC_VD_RAI2_before_request_to_HMI.lua
./test_scripts/Resumption/Handling_errors_from_HMI/033_Same_RPC_VD_RAI2_before_erroneous_response_from_HMI.lua
./test_scripts/Resumption/Handling_errors_from_HMI/034_Same_RPC_VD_RAI2_after_erroneous_response_from_HMI.lua
./test_scripts/Resumption/Handling_errors_from_HMI/038_Another_RPC_VD_RAI2_before_request_to_HMI.lua
./test_scripts/Resumption/Handling_errors_from_HMI/039_Another_RPC_VD_RAI2_before_erroneous_response_from_HMI.lua
./test_scripts/Resumption/Handling_errors_from_HMI/040_Another_RPC_VD_RAI2_after_erroneous_response_from_HMI.lua
./test_scripts/Resumption/Handling_errors_from_HMI/044_Another_RPC_VD_delayed_response_from_HMI.lua
```

jacobkeeler, 2020-12-10:
Closing with the merge of #3590

PR #3590 (2020-12-10): Handle omitted parameter permissions properly in OnVehicleData notification

Fixes #3577 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Run ATF tests listed in https://github.com/smartdevicelink/sdl_core/issues/3577#issuecomment-729083063

### Summary
Allow all parameters in OnVehicleData if parameters field is omitted

### Changelog
##### Bug Fixes
* Allow all parameters in OnVehicleData if parameters field is omitted

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYV [... cut]
closed 2020-12-10 by jacobkeeler