smartdevicelink/sdl_core #3580: Revoked app can be activated via device consent in EXTERNAL_PROPRIETARY

jacobkeeler, 2020-11-18:
### Bug Report
In EXTERNAL_PROPRIETARY policy mode, if an app is revoked, it can still be activated by providing consent for the device. This activation results in the app being brought to the foreground whenever it is reconnected as well.

##### Reproduction Steps
Using SDL HMI
1. app1 is revoked and connected, but device is not consented
2. Select app1 from the app list, accept consent prompt for device, HMI sends `SDL.OnAllowSDLFunctionality(true)`

##### Expected Behavior
app1 is not activated via `BC.ActivateApp` from SDL Core

##### Observed Behavior
SDL Core sends `BC.ActivateApp` to HMI, HMI brings app to foreground (although app1 still has `hmiLevel: NONE`). `BC.ActivateApp` is also sent to the HMI when the app reconnects.

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: 7.0.0
* Testing Against: Android Test Suite app

aderiabin, 2020-12-08:
Current issue is not regression one. It can be reproduced on SDL versions less than 7.0.

ShobhitAd, 2021-01-15:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3592

PR #3592 (2021-01-11): Fix/Revoked App Consent Activation
Fixes #3580 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Manual Testing with Test suite app

### Summary
- Add permissions check in OnAllowSDLFunctionalityNotification for application revoked case

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2021-01-15 by ShobhitAd