smartdevicelink/sdl_core #3603: SDL is unable to stream different audio files

dboltovskyi, 2020-12-15:
### Bug Report
SDL is unable to stream different audio files. 
1st file can be streamed successfully however there are errors with subsequent ones.

##### Reproduction Steps
1. Make sure `AudioStreamConsumer  = socket` in .INI file
2. Register and activate NAVI app
3. Start Audio service
4. Start streaming 1st file and wait until streaming is finished
5. Start streaming 2nd file (or the same file one more time)

For steps 5 and 6 check file is playing correctly using GStreamer:
```
gst-launch-1.0 souphttpsrc location=http://0.0.0.0:5080 ! audio/x-raw,format=S16LE,rate=16000,channels=1 ! pulsesink
```

##### Expected Behavior
File has played successfully with appropriate duration.

##### Observed Behavior
File can't be played due to errors observed in GS:

1)
```
ERROR: from element /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0:
streaming stopped, reason error (-5)
```

2)
```
ERROR: from element /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0: Could not establish connection to server.
Additional debug info:
gstsouphttpsrc.c(1375): gst_soup_http_src_parse_status (): /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0:
Could not connect: Connection refused (4), URL: http://0.0.0.0:5080, Redirect to: (NULL)
ERROR: pipeline doesn't want to preroll.
```

Only restart of the Audio service helps to play 2nd file

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (2e5c3ab)
* Testing Against: SPT

##### Test Case, Sample Code, and / or Example App
- SDL Log: [SmartDeviceLinkCore_socket.log.tar.gz](https://github.com/smartdevicelink/sdl_core/files/5697664/SmartDeviceLinkCore_socket.log.tar.gz)

dboltovskyi, 2020-12-15:
Issue is a regression one and starts reproducing within fix https://github.com/smartdevicelink/sdl_core/pull/3488 which was part of `7.0.0` release.
Issue is not reproduced at `6.1.1` release

JackLivio, 2021-01-25:
@dboltovskyi Does this issue also impact video streaming?

dboltovskyi, 2021-01-25:
@JackLivio Yes, we have recently found out Video streaming is also impacted.
Testing of the issue with GStreamer is not easy, since it has to be restarted for each start streaming attempt.

However the following tools can be used to verify the issue:
 - Web-HMI from https://github.com/smartdevicelink/sdl_hmi/pull/391. Allows to check `pipe` streaming type and only Video.
 - SharpHMI. Allows to check both Video and Audio with `socket` streaming type.

JackLivio, 2021-01-25:
@dboltovskyi I believe this sdl hmi pr allows for testing video using a socket setup? https://github.com/smartdevicelink/sdl_hmi/pull/391

dboltovskyi, 2021-01-25:
@AKalinich-Luxoft Could you please assist with the question [3603#issuecomment-767136074](https://github.com/smartdevicelink/sdl_core/issues/3603#issuecomment-767136074)

AKalinich-Luxoft, 2021-01-26:
@dboltovskyi @JackLivio yes, this HMI should work with both socket and pipe transports

AKalinich-Luxoft, 2021-01-28:
Was fixed in scope of #3468

closed 2021-02-22 by JackLivio