smartdevicelink/sdl_core #3606: [0328] Modernize Ubuntu Support V20 LTS

jordynmackool, 2020-12-16:
**Proposal:** [Modernize Ubuntu Support V20 LTS](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0328-modernize-ubuntu-support-v20.md)

> This proposal would like to provide support for Ubuntu 20 LTS and bump the minimum required version of Ubuntu to Ubuntu 18 LTS. Currently Ubuntu 18 is the recommended platform and Ubuntu 16 is the minimum required version.

>These changes will mainly affect the build process for SDL Core and ATF.

**Review:** https://github.com/smartdevicelink/sdl_evolution/issues/1103

**Steering Committee Decision:** 
The Steering Committee voted to fully accept this proposal.

JackLivio, 2021-04-13:
Ubuntu 20 support is implemented in the 7.1.0 release but this proposal is not fully implemented because the proposal mentions dropping support for Ubuntu 16. Dropping support for a platform requires a major version change for core and this issue will be fully resolved when it is decided Core should have a major version change on a release.

ShobhitAd, 2021-07-09:
> Ubuntu 20 support is implemented in the 7.1.0 release but this proposal is not fully implemented because the proposal mentions dropping support for Ubuntu 16. Dropping support for a platform [...]

Ubuntu 16 support dropped in https://github.com/smartdevicelink/sdl_core/pull/3727

PR #3727 (2021-07-09): Feature/Update Minimum Supported Version to Ubuntu 18
Fixes #3606 

This PR is **not ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
ATF Regression

### Summary
- Remove older SSL ciphers
- Remove references to ubuntu 16
- Update `clang-format` version to 8

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2021-07-09 by ShobhitAd