smartdevicelink/sdl_core #3618: Rare crash during UnsubscribeVehicleData

jacobkeeler, 2021-01-25:
### Bug Report
Occasionally, unsubscribing from vehicle data can cause a SIGSEGV crash

##### Reproduction Steps
1. Run [001_UnsubscribeVD_Success.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/API/VehicleData/UnsubscribeVehicleData/001_UnsubscribeVD_Success.lua)

##### Expected Behavior
The test finishes with SUCCESS result code

##### Observed Behavior
The test sometimes finishes with ABORTED result code

##### OS & Version Information
* OS/Version: Ubuntu 18.04.5 LTS
* SDL Core Version: develop
* Testing Against: ATF (develop branch)

##### Test Case, Sample Code, and / or Example App
[001_UnsubscribeVD_Success.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/API/VehicleData/UnsubscribeVehicleData/001_UnsubscribeVD_Success.lua)

jacobkeeler, 2021-01-25:
Per https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2488#issuecomment-756867744, this issue seems to appear only on the latest updates of Ubuntu 18.04, it does not a appear to be reproducible in 16.04 or earlier versions of 18.04

dboltovskyi, 2021-01-25:
@jacobkeeler Please advise is there plan for PMs to fix this issue?

jacobkeeler, 2021-01-25:
@dboltovskyi Just submitted a fix, see #3619

dboltovskyi, 2021-01-26:
@jacobkeeler Thank you for the fix

jacobkeeler, 2021-01-29:
closed via #3619

PR #3619 (2021-01-29): Fix invalid uses of erase while iterating through sets
Fixes #3618 and #3621 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Run all Vehicle Data ATF tests, Resumption Data Error Handling ATF tests, and unit tests

### Summary
Fix use of erase while iterating through set in `UnsubscribePendingVehicleData`, iterating through a copy of the set instead. Fix similar issues in `UnsubscribeFromInteriorVehicleDataOfType` and `DeleteApplicationData`.

### Changelog
##### Bug Fixes
* Fix several segfaults which occurred due to elements being erased from  [... cut]
closed 2021-01-29 by jacobkeeler