smartdevicelink/sdl_core #3620: SDL does not respond with NACK to start RPC service request with invalid data

dboltovskyi, 2021-01-26:
### Bug Report
SDL does not respond with NACK to start RPC service request with invalid data

##### Reproduction Steps
1. Start SDL, HMI, connect mobile device
2. App tries to start RPC service over 5th SDL protocol with invalid data in bson payload

##### Expected Behavior
SDL does:
 - send 'OnServiceUpdate' notification to HMI with 'REQUEST_REJECTED'
 - respond with `NACK` to start service request with appropriate `reason`

##### Observed Behavior
SDL does:
 - send 'OnServiceUpdate' notification to HMI with 'REQUEST_REJECTED'
 - not send `StartServiceNack` to mobile app

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (61ea4ef)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: 
 - Report and Logs:

LitvinenkoIra, 2021-02-08:
@VladSemenyuk is working on it

iCollin, 2021-03-24:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3647

PR #3647 (2021-03-24): Fix/sdl 3620 SDL does not respond with NACK to start RPC service request with invalid data
Fixes #3620

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF, Unit tests

### Summary
Fix memory leak with session after Start RPC service NACK.
Fix OnServiceUpdate sending after Start RPC service with invalid data in bson payload for protected mode.
Fix NACK sending for Start any service with non-existent session id.

##### Bug Fixes
**Start RPC service with invalid data in bson payload**
not send StartServiceNack to mobile app
session added in SDL Conn [... cut]
closed 2021-03-24 by iCollin