smartdevicelink/sdl_core #3621: FAIL occurs in ApplicationHelperTest when unit test is performed with SDL Core 7.0.0

Yuki-Kanadome-Nexty, 2021-01-26:
### Bug Report
FAIL occurs in ApplicationHelperTest when unit test is performed with SDL Core 7.0.0

##### Reproduction Steps
1. Clone SDL Core 7.0.0
>git clone https://github.com/smartdevicelink/sdl_core -b 7.0.0
2. Build according to the procedure (https://smartdevicelink.com/en/guides/core/getting-started/install-and-run/)
  Run CMake below
>cmake -DENABLE_TESTS=ON ../sdl_core
3. Test execution after build completion
>make test

##### Expected Behavior
Pass all tests

##### Observed Behavior
The following FAIL occurs
Test log:
>The following tests FAILED:
>14 --application_manager_test (Failed)
FAIL information in application_manager_test.xml:
>ApplicationHelperTest - RecallApplicationData_ExpectAppDataReset
>unknown file
>C ++ exception with description "Mock function called more times than expected --returning default value.
>Function call: CreateDeleteUICommandRequest (0x707070706070706, 1, 4)
>The mock function has no default action set, and its return type has no default value set. "Thrown in the test body.

##### OS & Version Information
* OS/Version: Ubuntu 18.04.5 LTS
* SDL Core Version: 7.0.0

dboltovskyi, 2021-01-26:
Please notice possible fix is described in https://github.com/smartdevicelink/sdl_core/pull/3619#issuecomment-767212365

jacobkeeler, 2021-01-29:
closed via #3619

closed 2021-01-29 by jacobkeeler