smartdevicelink/sdl_core #3639: SDL build fails under Ubuntu 20.04 with default gcc 9.3.0 compiler

dboltovskyi, 2021-02-23:
### Bug Report
Within introduction of Ubuntu 20.04 support SDL should be able to be built with default compiler (gcc 9.3.0)
However there is an error during build.

Note: SDL build succeeded if compiler downgraded to gcc 8.3.0

##### Reproduction Steps
1. Setup clean Ubuntu 20.04 environment
2. Clone `sdl_core` repository
3. Create build folder and try to build project with the commands:
```
  cmake ../sdl_core && make install-3rd_party_logger && make install -j4
```

##### Expected Behavior
SDL is built successfully

##### Observed Behavior
There is a build error:
```
/home/developer/sdl/b/3rd_party/include/boost/beast/core/impl/file_stdio.ipp: In member function ‘void boost::beast::file_stdio::seek(uint64_t, boost::beast::error_code&)’:
/home/developer/sdl/b/3rd_party/include/boost/beast/core/impl/file_stdio.ipp:185:15: error: comparison of integer expressions of different signedness: ‘uint64_t’ {aka ‘long unsigned int’} and ‘long int’ [-Werror=sign-compare]
  185 |     if(offset > (std::numeric_limits<long>::max)())
      |        ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [src/components/hmi_message_handler/CMakeFiles/HMIMessageHandler.dir/build.make:89: src/components/hmi_message_handler/CMakeFiles/HMIMessageHandler.dir/src/mb_controller.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2425: src/components/hmi_message_handler/CMakeFiles/HMIMessageHandler.dir/all] Error 2
```

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (d8cd806)
* Testing Against: N/A

##### Test Case, Sample Code, and / or Example App
 - Docker environment: [20.04_gcc_9.tar.gz](https://github.com/smartdevicelink/sdl_core/files/6032294/20.04_gcc_9.tar.gz)

dboltovskyi, 2021-02-24:
After analysis it was found out that if `THIRD_PARTY_INSTALL_PREFIX` and `THIRD_PARTY_INSTALL_PREFIX_ARCH` environment variables are not defined build succeeded.

Usually we define these ones in order to have all 3rd party libraries built into defined folder (not to default `/usr/local/lib`)

jacobkeeler, 2021-03-04:
@dboltovskyi can you verify if #3651 fixes this issue on your end? I was able to install using a custom install directory without any issues using those changes

dboltovskyi, 2021-03-04:
@jacobkeeler Sure. Thank you for the fix. I have already started testing it. Will get back with results.

jacobkeeler, 2021-03-04:
@dboltovskyi It's looking like some of the boost websocket functionality (including the HMI message handler) is broken in that PR, so we're debugging that at the moment. That said, it would still be helpful to know if the CMake changes are correct for fixing the build on your end

dboltovskyi, 2021-03-05:
@jacobkeeler Please notice fix you provided works fine on our environments

jacobkeeler, 2021-03-05:
closing with the merge of #3651

PR #3651 (2021-03-05): Update and fix Boost installation

Fixes #3639 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Run normal Core build, Boost 1.72.0 should be installed and the build should complete without any issues

### Summary
Updates the default Boost library installation to 1.72.0 and adds clean step to prevent old installations from breaking the build.

### Changelog
##### Enhancements
* Updates Boost library installation to 1.72.0

##### Bug Fixes
* Cleans Boost library installation before updating to prevent build errors. Also set installation paths prop [... cut]
closed 2021-03-05 by jacobkeeler