smartdevicelink/sdl_core #3640: SDL crashes while sending OnSystemRequest in case if RPC service is protected and SDL protocol version is 3rd or higher

dboltovskyi, 2021-02-25:
### Bug Report
This is a continuation of original issue which is described in https://github.com/smartdevicelink/sdl_core/issues/2922 and had been fixed in scope of Release 6.0.
Current issue addresses the same scenario but version of SDL protocol is >= 3

##### Reproduction Steps
1. Register new app with SDL protocol version >= 3
=> Policy Update is started
2. Switch RPC service to protected mode
=> SDL continues PTU sequence
=> SDL->App: OnSystemRequest(with snapshot)

##### Expected Behavior
SDL proceed with OnSystemRequest notification successfully

##### Observed Behavior
SDL crashes with error:
```
smartDeviceLinkCore: /home/developer/sdl/sdl_core/src/components/security_manager/src/ssl_context_impl.cc:470:
virtual bool security_manager::CryptoManagerImpl::SSLContextImpl::Encrypt(const uint8_t*, size_t, const uint8_t**, size_t*):
Assertion `(len == static_cast<size_t>(read_size))' failed.
```

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (d8cd806)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script:
 - Report and Logs:

dboltovskyi, 2021-02-25:
SDL needs to follow maximum size of protected records. According to spec this is 16384 bytes:
https://tools.ietf.org/id/draft-ietf-tls-record-limit-01.html#:~:text=TLS%20versions%201.2%20and%20earlier,expansion%20is%20only%2016%20octets

jacobkeeler, 2021-03-05:
Closing with the merge of #3652

PR #3652 (2021-03-05): Fix SSL max block size logic

Fixes #3640 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Run test script provided in #3640 

### Summary
Fixes `get_max_block_size` logic when the cipher name provided is unknown. Also adds the cipher name used in the provided test to the list of known ciphers.

### Changelog
##### Bug Fixes
* Fixes `get_max_block_size` logic when the cipher name provided is unknown. 
* Adds missing cipher name to the list of known ciphers.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VB [... cut]
closed 2021-03-05 by jacobkeeler