smartdevicelink/sdl_core #3660: ResumeController considers WARNINGS with message as reason to fail resumption

collinmcq, 2021-03-11:
### Bug Report
An HMI response with "message", regardless of result code, will be considered by the resumption controller as unsuccessful, causing resumption to fail and these responses to not be reverted.

##### Reproduction Steps
1. Start SDL Core and SDL HMI
2. Connect Mobile App
3. Mobile App Sends an AddCommand with a DYNAMIC image that does not exist on the file system
4. Mobile App reconnects using the last hashID it received

##### Expected Behavior
SUCCESS response to RAI, AddCommand resumed

##### Observed Behavior
RESUME_FAILED response to RAI, AddCommand that got WARNINGS was not reverted

##### Test Case, Sample Code, and / or Example App
ATF Test with slight modifications:
```
diff --git a/test_scripts/Resumption/Handling_errors_from_HMI/027_resume_all_data_with_one_error_rpc_unexpected_disconnect_different_resultCodes.lua b/test_scripts/Resumption/Handling_errors_from_HMI/027_resume_all_data_with_one_error_rpc_unexpected_disconnect_different_resultCodes.lua
index 56891c56..b2bf4801 100644
--- a/test_scripts/Resumption/Handling_errors_from_HMI/027_resume_all_data_with_one_error_rpc_unexpected_disconnect_different_resultCodes.lua
+++ b/test_scripts/Resumption/Handling_errors_from_HMI/027_resume_all_data_with_one_error_rpc_unexpected_disconnect_different_resultCodes.lua
@@ -66,7 +66,7 @@ local function reRegisterApp(pAppId, pErrorCode)
     end)
   common.getHMIConnection():ExpectRequest("UI.AddSubMenu",common.resumptionData[pAppId].addSubMenu.UI)
   :Do(function(_, data)
-      common.getHMIConnection():SendError(data.id, data.method, pErrorCode, "info message")
+      common.getHMIConnection():SendError(data.id, data.method, pErrorCode)
     end)
 
   common.resumptionFullHMILevel(pAppId)
```

dboltovskyi, 2021-03-11:
@iCollin Please notice `WARNINGS` code is a successful one and needs to be sent from HMI with `SendResponse()` function which creates successful response structure. This result code is covered in [002_successful_resumption_other_successful_result_codes](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Resumption/Handling_errors_from_HMI/002_successful_resumption_other_successful_result_codes.lua) script.

From other hand `SendError()` creates an erroneous structure, so the whole message is treated by SDL as erroneous one, and hence SDL fails resumption. 

As for https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2529 it looks like `TRUNCATED_DATA` is a successful one and that's correct - it's need to be removed from the erroneous list. Though it may be need to be added to `002` script.

collinmcq, 2021-03-11:
@dboltovskyi thank you, I have made changes to 002 script to include `TRUNCATED_DATA`.

closed 2021-03-11 by JackLivio