smartdevicelink/sdl_core #3664: Unable to establish secure connection on Ubuntu 20.04

dboltovskyi, 2021-03-17:
### Bug Report
Unable to establish secure connection on Ubuntu 20.04. Though it can be done successfully on Ubuntu 18.04.
Same SDL commit, SPT version, settings and certificates.
openssl version:
- Ubuntu 20.04: `1.1.1f`
- Ubuntu 18.04: `1.1.1`

##### Reproduction Steps
1. Make sure all required certificates are defined properly
2. Start SDL, HMI, connect mobile device, start SPT
3. Register new App
4. Enable secure RPC service

##### Expected Behavior
Handshake sequence is finished successfully and RPC service is switched to protected mode

##### Observed Behavior
Handshake sequence has failed and RPC service hasn't switched to protected mode

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (0df6608)
* Testing Against: SPT

##### Test Case, Sample Code, and / or Example App
 - Ubuntu 20.04 SDL log: [u20_SmartDeviceLinkCore.log.tar.gz](https://github.com/smartdevicelink/sdl_core/files/6153888/u20_SmartDeviceLinkCore.log.tar.gz) (FAIL)
 - Ubuntu 18.04 SDL log: [u18_SmartDeviceLinkCore.log.tar.gz](https://github.com/smartdevicelink/sdl_core/files/6153887/u18_SmartDeviceLinkCore.log.tar.gz) (SUCCESS), provided for comparison

LitvinenkoIra, 2021-03-17:
Luxoft team is going to provide a fix.

jordynmackool, 2021-03-31:
@LitvinenkoIra please advise when the fix for this issue will be ready for Livio review?

dboltovskyi, 2021-03-31:
@jordynmackool Please notice fix for this issue is in our internal team review. There is a plan to publish it for PM review until tomorrow.

jacobkeeler, 2021-04-06:
closing with the merge of #3681

PR #3681 (2021-04-06): Configure security level in ini file
Fixes  #3664

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
Check secure connection with SyncProxyTester app 

### Summary
In Ubuntu 20.04, the OpenSSL 1.1.1f library has been modified to use Security Level 2 by default.
With this PR it becomes possible to change the Security Level in .ini file for establish secure connection with some applications that do not support this level.

In details about levels from openssl doc (https://www.openssl.org/docs/man1.1.0/man3/SSL_CTX_get_security_level.html):

Level  [... cut]
closed 2021-04-06 by jacobkeeler