smartdevicelink/sdl_core #3666: SDL saves wrong custom_vehicle_data_mapping_url version in the DB and sends it in the next ignition cycle

GetmanetsIrina, 2021-03-17:
### Bug Report
SDL saves the wrong custom_vehicle_data_mapping_url version in DB and sends it via GetPolicyConfigurationData in the next ignition cycle

##### Presonfdition
1. sdl_preloaded_pt.json file has version="0.0.0" in custom_vehicle_data_mapping_url structure
2. SDL and HMI are started

##### Reproduction Steps
1. Check custom_vehicle_data_mapping_url version in local DB
2. Perform IGN_OFF - IGN_ON
3. Go to HMI main settings  → "Policies" → "Send GetPolicyConfigurationData request to SDL" → "Check Custom Vehicle Data Mapping Update Flow"

##### Expected Behavior
1. DB contains a value in custom_vehicle_data_mapping_url version that is equal to the value from the sdl_preloaded_pt.json file
2. In the second ignition cycle SDL sends GetPolicyConfigurationData response with actual custom_vehicle_data_mapping_url version (version="0.0.0")

##### Observed Behavior
1. SDL saves wrong value(1) for custom_vehicle_data_mapping_url into the DB
2. In the second ignition cycle SDL sends GetPolicyConfigurationData response with wrong custom_vehicle_data_mapping_url version (version="1")
3. SDL adds the custom_vehicle_data_mapping_url entry into the DB each time when an update of OEM data is triggered

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (c9fc559)
* Testing Against: HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop (41c804d), SPT: SPT: 20210305-Android (1076) 

##### Logs
[SDL log and DB.zip](https://github.com/smartdevicelink/sdl_core/files/6157127/SDL.log.and.DB.zip)

jordynmackool, 2021-03-18:
Hi @GetmanetsIrina, does Luxoft plan on submitting a fix for this reported issue?

GetmanetsIrina, 2021-03-22:
@jordynmackool, This is not a regression issue so Luxoft can look at it after 7.1 release will be released. Livio can begin investigating this issue for now.

LitvinenkoIra, 2021-04-06:
Hi @jordynmackool, please advise if this fix will be included in the 7.1 release?

jordynmackool, 2021-04-06:
Hi @LitvinenkoIra, as this is noted to not be a regression issue, Livio will plan to investigate it following the 7.1 Release.

jacobkeeler, 2021-05-20:
Closing with the merge of #3704

PR #3704 (2021-05-20): Fix saving to endpoint properties table

Fixes #3666 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF PR in progress, test manually by checking `endpoint_properties` field in policy table

### Summary
`EndpointProperty.version` is defined as an optional value, but was being used as a mandatory one. Also makes `service` property unique in `endpoint_properties` to prevent duplicate entries.

### Changelog
##### Bug Fixes
* Fixes incorrect saving of endpoint properties and restricts duplicate saving

### CLA
- [x] I have signed [the CLA](https:/ [... cut]
closed 2021-05-20 by jacobkeeler