smartdevicelink/sdl_core #3667: SDL does not send revert RC.SetGlobalProperties requests to HMI during revert of resumption in case of erroneous response type

VjKlepikov, 2021-03-18:
### Bug Report
SDL does not send revert RC.SetGlobalProperties requests to HMI during revert of resumption in case of erroneous response type

### Precondition:
    1. Start SDL and HMI.
    2. Connect mobile device via any transport.
    3. Mobile application is registered and activated.
    4. Mobile application successfully sends SetGlobalProperties(UI, TTS, RC)
    ...
     SetGlobalProperties { "userLocation":{"grid": {"col":2, "level":2,"row":2, "colspan":1,"levelspan":1, "rowspan":1 }
    5. On HMI go to "Setting -> RPC Control -> App name -> UI.SetGlobalProperties
       Set ERROR resultCode (ex. GENERIC_ERROR ) from the drop-down list and click "Back" button
    6. Mobile application is unregistered with unexpectedDisconnect=true during transport disconnect
    7. Mobile application is registered with actual hashID
       Turn on transport on Mobile device
       SPT -> Menu -> Reset Connection
    8. [Interface(UI/VR/RC)].[ SetGlobalProperties ] is sent from SDL to HMI during resumption for application.

##### Reproduction Steps
HMI responds with an erroneous type of response ( ERROR resultCode ( ex. GENERIC_ERROR))
to UI.SetGlobalProperties and request from SDL

    1. SDL→HMI: UI.SetGlobalProperties ( params )
    2. SDL→HMI: TTS.SetGlobalProperties ( params )
    3. SDL→HMI: RC.SetGlobalProperties
    { "userLocation":{"grid":{"col":2, "level":2,"row":2, "colspan":1,"levelspan":1, "rowspan":1 }}} 

    4. HMI→SDL: TTS.SetGlobalProperties ( resultCode: SUCCESS )
    5. HMI→SDL: RC.SetGlobalProperties ( resultCode: SUCCESS )
    6. HMI→SDL: UI.SetGlobalProperties ( resultCode = ( ex. GENERIC_ERROR) )

##### Expected Behavior
-     Remove already restored data
-     Send set of revert <Rpc_n> requests to HMI (except the one related to <erroneous> response)
    1. SDL→HMI: TTS.SetGlobalProperties ( default params )
    2. SDL sends revert RC.SetGlobalProperties request with default driver seat location properties to HMI
         SDL→HMI: RC.SetGlobalProperties (
{ "userLocation":{"grid":{"col":0, "level":0,"row":0, "colspan":1,"levelspan":1, "rowspan":1 }}} 
Note: Mobile app received respond RegisterAppInterfaceResponse( success=true, result_code=RESUME_FAILED )

##### Observed Behavior
-     Remove already restored data
-     Send set of revert <Rpc_n> requests to HMI (except the one related to <erroneous> response)
    1. SDL→HMI: TTS.SetGlobalProperties ( default params )
    2. SDL does not send revert RC.SetGlobalProperties requests with default driver seat location properties  to HMI

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (c9fc559)
* Testing Against: https://github.com/smartdevicelink/sdl_hmi/tree/develop (41c804d), SPT: 20210305-Android (1076)

jordynmackool, 2021-03-18:
Hi @VjKlepikov, does Luxoft plan on submitting a fix for this reported issue?

ShobhitAd, 2021-03-22:
Hey @VjKlepikov @jordynmackool, I was able to reproduce the issue on my end and I'm working on a fix for it

ShobhitAd, 2021-03-25:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3673

PR #3673 (2021-03-25): Fix/Reset RC global properties on Failed Resumption
Fixes #3667 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Run issue reproduction steps using Test Suite/RPC Builder

### Summary
- Implemented ResetGlobalProperties request for RC properties
- Added functions `HasRCPropertiesReset` and `CreateRCResetGlobalPropertiesRequest` to match the behaviour for TTS and UI global properties

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2021-03-25 by ShobhitAd